OpenIntelWireless/itlwm #190: magicbook pro 2020 网卡9560 160mhz devId=0x02F0报错

sursamer, 2020-07-15:
2020-07-15 14:57:21.872238+0800 0x1300     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x3E9B
2020-07-15 14:57:21.872243+0800 0x1300     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.872264+0800 0x130f     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02A8
2020-07-15 14:57:21.872269+0800 0x130f     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.872296+0800 0x1310     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02AB
2020-07-15 14:57:21.872301+0800 0x1310     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.872403+0800 0x1311     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x0284
2020-07-15 14:57:21.872407+0800 0x1311     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.872426+0800 0x12ff     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x9B61
2020-07-15 14:57:21.872430+0800 0x12ff     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.872519+0800 0x1313     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02A3
2020-07-15 14:57:21.872523+0800 0x1313     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.872723+0800 0x1301     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x1903
2020-07-15 14:57:21.872727+0800 0x1301     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.873013+0800 0x1314     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02A4
2020-07-15 14:57:21.873043+0800 0x1303     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02F9
2020-07-15 14:57:21.873047+0800 0x1303     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.873078+0800 0x1314     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_attach
2020-07-15 14:57:21.873086+0800 0x1314     Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: PCIe capability structure not found!
2020-07-15 14:57:21.873174+0800 0x130b     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02C5
2020-07-15 14:57:21.873178+0800 0x130b     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.873204+0800 0x1309     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02E0
2020-07-15 14:57:21.873209+0800 0x1309     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.873341+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02F0
2020-07-15 14:57:21.873343+0800 0x1308     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02EA
2020-07-15 14:57:21.873347+0800 0x1308     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.873373+0800 0x1305     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x02EF
2020-07-15 14:57:21.873379+0800 0x1305     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.873778+0800 0x1302     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_match devId=0x1911
2020-07-15 14:57:21.873781+0800 0x1302     Default     0x0                  0      0    kernel: (itlwm) itlwm: free
2020-07-15 14:57:21.883671+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_attach
2020-07-15 14:57:21.883771+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: , (null)
2020-07-15 14:57:21.883787+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_prepare_card_hw
2020-07-15 14:57:21.883992+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: alloc contig
2020-07-15 14:57:21.965612+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: config ieee80211
2020-07-15 14:57:21.965614+0800 0x1317     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_create_thread lock
2020-07-15 14:57:21.965615+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) ieee80211_ifattach
2020-07-15 14:57:21.965617+0800 0x1317     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_create_thread unlock
2020-07-15 14:57:21.965618+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_channel_init
2020-07-15 14:57:21.965622+0800 0x1318     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_create_thread lock
2020-07-15 14:57:21.965623+0800 0x1317     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_create_thread lock
2020-07-15 14:57:21.965624+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state nstate=2, old_state=-1
2020-07-15 14:57:21.965625+0800 0x1318     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_create_thread unlock
2020-07-15 14:57:21.965626+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
2020-07-15 14:57:21.965627+0800 0x1319     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_thread schedule task
2020-07-15 14:57:21.965627+0800 0x1317     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_create_thread unlock
2020-07-15 14:57:21.965630+0800 0x1318     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_create_thread lock
2020-07-15 14:57:21.965631+0800 0x131a     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_thread schedule task
2020-07-15 14:57:21.965633+0800 0x1318     Default     0x0                  0      0    kernel: (itlwm) itlwm: taskq taskq_create_thread unlock
2020-07-15 14:57:21.965637+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: ieee80211_media_init
2020-07-15 14:57:21.965641+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_preinit
2020-07-15 14:57:21.965642+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_prepare_card_hw
2020-07-15 14:57:21.965666+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_start_hw
2020-07-15 14:57:21.965667+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_prepare_card_hw
2020-07-15 14:57:21.971331+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_apm_init
2020-07-15 14:57:21.971368+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: L1 Disabled - LTR Enabled
2020-07-15 14:57:21.971528+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_init_msix_hw
2020-07-15 14:57:21.971529+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_conf_msix_hw
2020-07-15 14:57:21.971556+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_run_init_mvm_ucode justnvm=1
2020-07-15 14:57:21.971558+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_load_ucode_wait_alive ucode_type=1
2020-07-15 14:57:21.972501+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: load firmware iwm-9000-34 done
2020-07-15 14:57:21.972505+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: Paging: paging enabled (size = 241664 bytes)
2020-07-15 14:57:21.972510+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_start_fw ucode_type=1
2020-07-15 14:57:21.972511+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_apm_init
[... cut ...]
2020-07-15 14:57:23.044746+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_conf_msix_hw
2020-07-15 14:57:23.044779+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_prepare_card_hw
2020-07-15 14:57:23.044783+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: attach succeed.
2020-07-15 14:57:23.044796+0800 0x1306     Default     0x0                  0      0    kernel: (itlwm) itlwm: attach to interface fail

zxystd, 2020-07-17:
link #127 .

closed 2020-07-17 by zxystd

sursamer, 2020-07-17:
@zxystd   使用了itlwmx之后 报错变了，我贴下日志
2020-07-17 10:01:59.344003+0800 0x2923a    Default     0x0                  0      0    kernel: (itlwmx) _start(struct kmod_info*, void*) has been invoked
2020-07-17 10:01:59.345865+0800 0x29250    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.346024+0800 0x2924c    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.346228+0800 0x29256    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.346668+0800 0x2925b    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.346699+0800 0x2924d    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.346786+0800 0x29253    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.346877+0800 0x2924a    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.347040+0800 0x2924e    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.347250+0800 0x2924b    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.347334+0800 0x29254    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.347853+0800 0x2925a    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.348064+0800 0x2925f    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.348174+0800 0x2925c    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.358741+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_prepare_card_hw
2020-07-17 10:01:59.358744+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-07-17 10:01:59.467196+0800 0x29262    Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-07-17 10:01:59.467911+0800 0x29263    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread lock
2020-07-17 10:01:59.467913+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) ieee80211_ifattach
2020-07-17 10:01:59.467914+0800 0x29263    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread unlock
2020-07-17 10:01:59.467914+0800 0x29264    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread lock
2020-07-17 10:01:59.467917+0800 0x29264    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread unlock
2020-07-17 10:01:59.467920+0800 0x29263    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread lock
2020-07-17 10:01:59.467920+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_channel_init
2020-07-17 10:01:59.467922+0800 0x29264    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread lock
2020-07-17 10:01:59.467923+0800 0x29265    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_thread schedule task
2020-07-17 10:01:59.467923+0800 0x29263    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread unlock
2020-07-17 10:01:59.467924+0800 0x29264    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_create_thread unlock
2020-07-17 10:01:59.467926+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_set_link_state nstate=2, old_state=-1
2020-07-17 10:01:59.467927+0800 0x29266    Default     0x0                  0      0    kernel: (itlwmx) itlwm: taskq taskq_thread schedule task
2020-07-17 10:01:59.467928+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_set_link_state LINK_STATE_IS_DOWN
2020-07-17 10:01:59.467931+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_media_init
2020-07-17 10:01:59.467935+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_prepare_card_hw
2020-07-17 10:01:59.467936+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-07-17 10:01:59.467952+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_start_hw
2020-07-17 10:01:59.467953+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_prepare_card_hw
2020-07-17 10:01:59.467954+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-07-17 10:01:59.473714+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : L1 Disabled - LTR Enabled
2020-07-17 10:01:59.473933+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_run_init_mvm_ucode
2020-07-17 10:01:59.473934+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_load_ucode_wait_alive
2020-07-17 10:01:59.474324+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: load firmware done
[... cut ...]
2020-07-17 10:02:00.778141+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_prepare_card_hw
2020-07-17 10:02:00.778144+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-07-17 10:02:00.778211+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: attach to interface fail
2020-07-17 10:02:00.790940+0800 0x29251    Default     0x0                  0      0    kernel: (itlwmx) itlwm: : can't establish interrupt