OpenIntelWireless/itlwm #192: Intel AC3165无线网卡Bug提交

Code-God1024, 2020-07-16:
连接不上Windows系统下开的WI-FI热点，我用了三台Windows电脑开热点测试，都不可以，不知道算不算BUG，提交一下。别的Wi-Fi类型可以正常连接。

Code-God1024, 2020-07-16:
这三台WI-FI的名称和密码中都不含有特殊字符

zxystd, 2020-07-17:
重新复现一下，取一下log,
log show --last boot | grep itlwm >~/Desktop/itlwm.log
另外Wi-Fi名称是什么？加密类型是什么？

Code-God1024, 2020-07-17:
> 重新复现一下，取一下log,
>  log show --last boot | grep itlwm &gt;~/Desktop/itlwm.log
>  另外Wi-Fi名称是什么？加密类型是什么？
>
>
>
>
> 重新复现一下，取一下log,
>  log show --last boot | grep itlwm &gt;~/Desktop/itlwm.log
>  另外Wi-Fi名称是什么？加密类型是什么？

今天我们这里断电了，要到晚上11点恢复，log晚点发给你。
WiFi1名称biubiubiu，密码fuck2see;
WiFi2名称jiu，密码erziyaoguai;
WiFi3名称ccc，密码880196XY;
加密都是wpa2personal;

Code-God1024, 2020-07-18:
_

[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/4940857/itlwm.log)
_

Code-God1024, 2020-07-18:
日志可以吗，不行的话我重新上传

Code-God1024, 2020-07-18:
> 重新复现一下，取一下log,
> log show --last boot | grep itlwm >~/Desktop/itlwm.log
> 另外Wi-Fi名称是什么？加密类型是什么？
日志可以吗，不行的话我重新上传

zxystd, 2020-07-18:
> > 重新复现一下，取一下log,
> > log show --last boot | grep itlwm >~/Desktop/itlwm.log
> > 另外Wi-Fi名称是什么？加密类型是什么？
> 
> 日志可以吗，不行的话我重新上传

下载半天下载不下来，发我邮箱吧1051244836@qq.com

zxystd, 2020-07-19:
[image]
跟网卡硬件没关系，是你的热点有问题，给的IE不符号标准。
[image]

closed 2020-07-22 by zxystd