OpenIntelWireless/itlwm #195: System becomes unresponsive

naveenkrdy, 2020-07-19:
Whenever my network doesnt work due to an issue on my ISP end. My macos system becomes unresponsive thus i have to disconnect from my wifi network or turn off my wifi to fix it.

**Expected behaviour**
When there is no response from the server. it should not make the macos system unresponsize.

AX200
10.15.6
[`9e89092`](https://github.com/OpenIntelWireless/itlwm/commit/9e890928f3db6fef2545eb4f7547e582a31bf3a8)

Regards !

zxystd, 2020-07-22:
any logs?
log show --last boot | grep itlwm >~/Desktop/itlwm.log

naveenkrdy, 2020-08-06:
> any logs?
> `log show --last boot | grep itlwm >~/Desktop/itlwm.log`

Hi !
will provide it once i get the same issue again.

Thanks !

closed 2020-08-15 by williambj1