OpenIntelWireless/itlwm #196: AC3165 kextclover加载失败，手动加载失败

codeyuan123, 2020-07-19:
2020-07-20 00:04:39.389947+0800 0xda       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV13IOEventSource' for vtable '__ZTV11IOTaskQueue' is out of date. Make sure your kext has been built against the correct headers.
2020-07-20 00:04:39.389957+0800 0xda       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV19IOEthernetInterface' for vtable '__ZTV15itlwm_interface' is out of date. Make sure your kext has been built against the correct headers.
2020-07-20 00:04:39.389966+0800 0xda       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV8OSObject' for vtable '__ZTV15pci_intr_handle' is out of date. Make sure your kext has been built against the correct headers.
2020-07-20 00:04:39.389974+0800 0xda       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV12IOUserClient' for vtable '__ZTV20ItlNetworkUserClient' is out of date. Make sure your kext has been built against the correct headers.
2020-07-20 00:04:39.389982+0800 0xda       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV20IOEthernetController' for vtable '__ZTV5itlwm' is out of date. Make sure your kext has been built against the correct headers.
2020-07-20 00:04:39.389991+0800 0xda       Default     0x0                  0      0    kernel: kxld[com.zxystd.itlwm]: The super class vtable '__ZTV8OSObject' for vtable '__ZTV8CTimeout' is out of date. Make sure your kext has been built against the correct headers.
2020-07-20 00:04:39.392325+0800 0xda       Default     0x0                  0      0    kernel: Can't load kext com.zxystd.itlwm - link failed.
2020-07-20 00:04:39.396954+0800 0xda       Default     0x0                  0      0    kernel: Failed to load executable for kext com.zxystd.itlwm.
2020-07-20 00:04:39.397539+0800 0xda       Default     0x0                  0      0    kernel: Kext com.zxystd.itlwm failed to load (0xdc008016).
2020-07-20 00:04:39.397543+0800 0xda       Default     0x0                  0      0    kernel: Failed to load kext com.zxystd.itlwm (error 0xdc008016).
2020-07-20 00:04:39.398780+0800 0xda       Default     0x0                  0      0    kernel: Couldn't alloc class "itlwm"
2020-07-20 00:05:10.157641+0800 0x20e      Error       0x0                  44     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext id com.zxystd.itlwm not found; removing personalities from kernel.

Goshin, 2020-07-19:
显然编译这个kext的系统跟你的系统版本不一样，所以加载不了。

codeyuan123, 2020-07-20:
> 显然编译这个kext的系统跟你的系统版本不一样，所以加载不了。

感谢，已经重新编译了，完美使用

closed 2020-07-20 by williambj1