OpenIntelWireless/itlwm #199: Transmitting big data causes performance issues

rdmitry0911, 2020-07-22:
Hi guys, thank you for the driver and for you efforts. I'm trying to use my laptop with Cannon Point-LP CNVi wifi and latest itlwm driver under 10.15.6 as a KODI DLNA server. It doesn't work fine. Even if I put my laptop close to the AP, a client constantly stuck. At the same time I have 20Mbs download and 10Mbps upload sped according to OOKLA speedtest.  Running the same KODI DLNA server under Windows or Linux on the same laptop and with the same AP doesn't make any problems to the client. It means something is wrong in the driver or in its configuration. 

Regards, Dmitry

metro2450, 2020-08-25:
It looks like a typical upload speed issue.  I am using AX200 on my ThinkPad T460 and have no problem on download (Wireless Router is shown 144.44Mbps connection on 5GHz), however upload test is only 10Mbps.

isuke01, 2020-11-16:
Actually I'm facing same issue. DL speed goes down after some time.
If I try to download anything else both downloads again goes fast for some time then both slow down to 100-200kb/s. 
Speed test  by Ookla shows Download about 47Mbps, upload 10Mbps so in test it looks fine, but not in real usage.
For tests I used Chrome browser, Safari browser. 
Also I did test exact same files in the same time on other macDevice (macMini) - no issues at all.

My full internet speed is about 50 Mbps.

Device: Huawei Matebook X Pro 2018
Processor:   Core i7-8550U Processor @ 1.8 GHz
itlwm: 1.2 ALFA (release from 11h +- ago), 1.1.0 STABLE
Boot: OC 0.6.3
Big Sur 11.0.1
WiFi card: Intel Dual Band Wireless-AC 8265/8275

mehmetyukcell, 2020-11-16:
> Actually I'm facing same issue. DL speed goes down after some time.
>
> If I try to download anything else both downloads again goes fast for some time then both slow down to 100-200kb/s.
>
> Speed test [...]

Is AirDrop running on your system with AirPortitlwm 1.2

isuke01, 2020-11-16:
@mehmetyukcell 
Hey, sorry I was in edit mode, actually issue is the Dual-band WIFI setting. In standard mode wifi there is no issue.

isuke01, 2020-11-17:
Logs are from WiFi  and kext while I swapped to dual-band WI-FI ,first I was auto connected to another WIFI from prev test.
I run some file to download, speed from fast went down.
I hope I get correct logs. If needed I can do more texts / another logs.

I used during this test build  6b22398 1.2.0 Debug alfa. (But exact same issue I can see in 1.1.0 release)

[wifi_logs.zip](https://github.com/OpenIntelWireless/itlwm/files/5550396/wifi_logs.zip)

williambj1, 2021-02-25:
Should be much better with recent versions, request a reopen if that is not the case.

closed 2021-02-25 by williambj1