OpenIntelWireless/itlwm #200: Putting a laptop to a weak signal zone causes problems

rdmitry0911, 2020-07-22:
Hi guys, thank you for the driver and for you efforts. I have a laptop with Cannon Point-LP CNVi wifi and the latest itlwm driver under 10.15.6 The driver works fine when it is close to the AP. When I put it behind a wall, it looks like it is connected, but in fact it is not. I can not even ping a router. I have to disable and reenable wifi in Heliport to make it work. After reenabling wifi I have the connection, but the speed is very slow (about 30-50 Kb/s). From the same point under Windows and Linux I have reduced, but acceptable speed of about 10Mb/s

Regards, Dmitry

zxystd, 2020-07-24:
sorry i can't improve the signal, it depends on the wifi card. thanks.

rdmitry0911, 2020-07-24:
You didn't get the message. The signal strength is OK as far as windows and linux systems on the same laptop with the same wifi card work fine. Only macos with your driver has problems.

williambj1, 2020-07-24:
Drives for Windows and Linux are written by people who work in Intel, you can ask them to write a driver for macOS

rdmitry0911, 2020-07-24:
Hi guys, look, I didn't want to abuse you. Sorry, if I did. What I did want is to help you to make the driver as good as it is in linux and windows. I supposed, that this is the right place to put here all the problems your driver has while drivers in other operating systems don't have. Again, I respect your efforts and I like the driver you are hard working on.

williambj1, 2020-07-24:
Since most of the code of this project is ported from OpenBSD, I guess you could install it on a USB drive to verify whether it’s an issue of itlwm or the ported mac layer. It’s really hard to tell based on your given information and we can not reproduce this issue on any of our setups. Hope you understand the situation.

Edoardo001, 2020-07-24:
Don't know  if  ca be helpful, on 9560 i've experience less signal than windows. For me it's perfect as it  is, I never thought it was possible that all this would happen.

williambj1, 2020-07-24:
@Edoardo001 Could you please specify how did you notice the signal was poorer?

Edoardo001, 2020-07-24:
> @Edoardo001 Could you please specify how did you notice the signal was poorer?

If I do a Ookla speed test in my bathroom with windows I get 48megabits and I'm able to sift internet

With osx I can get only 4/5 mega with some problems to surfing internet (like that the 4/5 mega it's not real)

If can be helpful I can check WiFi dB signal. If not it's not problem, maybe this can avoid me work when I'm at poop, and this can be great 😂

williambj1, 2020-07-24:
I think that’s more related to the use of 802.11ac VS 802.11n; speed difference doesn’t represent signal strength difference. You can check heliport’s option menu and observe the RSSI value.

williambj1, 2020-07-24:
You can read this article for more information 

** 802.11ac vs 802.11n Range**
> All this means the performance of 802.11ac is maintained far better at long range than 802.11n. Peak performance may be tripled, but at range 5-10x the speed benefits are not unusual and this is where 802.11ac comes into its own.

https://www.forbes.com/sites/gordonkelly/2014/12/30/802-11ac-vs-802-11n-wifi-whats-the-difference/#2c35941a3957

Edoardo001, 2020-07-24:
> I think that’s more related to the use of 802.11ac VS 802.11n; speed difference doesn’t **always** represent signal strength difference. You can check heliport’s option menu and observe the RSSI value.

Will check, maybe it's just a difference between 802.11ac and 802.11n. Thanks for the link, will read. Use technology it's nice, knowing how technology that we are using, works, it's awesome

ataberkw, 2020-08-26:
I got 9560 and I have the exact same observation. Signal 2/5 is enough for Windows to surf on web stable. On the other side itlwm 2/5 cant even load website. I appreciate your work.

ataberkw, 2020-08-26:
One more thing. When I extend 2/5 wifi with my phone, 5/5 is still unusable. I understand you cant improve low wifi with hotspot but just wanted to give some little more information.