OpenIntelWireless/itlwm #201: [Feature Request] Map Funktion Key to Toggle WiFi/HeliPort

2eno, 2020-07-22:
Hello There,
I hope this is not the wrong place for this kind of request!
I just wondered whether it's possible to implement a feature so you can map a custom key to toggle wifi on and off.
Many laptops have a special "function key" to do this and as far as I know till now its not possible to use it on Hackintosh.
With this driver, and the control through an external App: "HeliPort", I thought it should be technically possible to implement a feature to support it?
Thanks for reading!

williambj1, 2020-07-23:
Theoretically, this would be possible, but it requires code changes in both VoodooPS2 and itlwm. Maybe @Goshin knows more about it.

1Revenger1, 2020-07-23:
Atleast for a lot of Thinkpads, they are EC queries. Though, normally what we do for backlight brightness is have them send PS2 key codes that correspond to brightness up and down, since those are keycodes that macOS and VoodooPS2 recognize.

1Revenger1, 2020-07-23:
Another approach is something similar to what [Thinkpad Assistance](https://github.com/MSzturc/ThinkpadAssistant) does, and use ACPI patches to send PS2 commands for Function keys above 12. We could capture those Function keys in HeliPort and show an icon when Wifi is enabled/disabled which may be a better option.

junaedahmed, 2020-07-23:
https://github.com/MSzturc/ThinkpadAssistant/issues/23#issue-627413413

Stevemoretz, 2020-08-16:
How about a few command that we could run in the command line to turn on, turn off, and get the status of wifi?Then we can use Automator service and put the commands there and then assign hotkey.Even tough the hotkey obviously wouldn't be fn + something but anyways, it's still great a lot of people included me don't even have a wifi button to press with fn.So why over complicate things for the start?

Goshin, 2020-08-16:
I just added a service in HeliPort to toggle WiFi.

After updating the app (remember to install it in `Applications`) and rebooting, you will find a service named `HeliPort: Toggle WiFi` in `System Preferences > Keyboard > Shortcuts > Services > General`, then you can assign a keyboard shortcut for it.

[image]

For Fn hotkeys based on EC queries, they are not able to be directly captured by the OS so it is out of the scope of itlwm and HeliPort. Luckily many laptop models send Fn hotkeys via WMI events. I have implemented a WMI driver months ago, with which you don't have to patch your DSDT to convert them to PS/2 key code. Check out the repo [here](https://github.com/Goshin/VoodooWMI).

shervintravel, 2020-08-25:
# Developer 
**DanPattaya**
**[Official Site](https://dandeveloper.carrd.co)**

closed 2020-09-19 by williambj1