OpenIntelWireless/itlwm #204: [Bug] When I use AirPods/AirPods Pro link with bluetooth, wi-fi signal break down

jiaogger, 2020-07-23:
**Describe the bug**
When I use AirPods/AirPods Pro link with bluetooth, wi-fi signal break down. Is there any interference orcrosstalk between the Wi-Fi signal and Bluetooth signal (I'm a hackintosh user)？

**To Reproduce**
Steps to reproduce the behavior:
1. linked wifi with heliport 
2. linked AirPods/AirPods Pro with bluetooth
3. open up a music APP like Spotify
4. the wifi signal break down

**Expected behavior**
can use airpods/AirPods Pro normally without the issue on wifi signal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS Version: Mac 10.15.5
 - Software Version: intel 9560

**Additional context**
Add any other context about the problem here.

zxystd, 2020-07-24:
no airpods, anyone can help to test and reproduce?

M69W, 2020-07-24:
I will test it tonight.
what I focus on? just only wifi signal?

almenjonatan, 2020-07-24:
No airpods, but blutooth headphones, same issue. Wifi becomes really slow or break down completely.

Catalina 10.15.6
Xiaomi Notebook Pro, intel 8260

Heli and itlwm version 0.7xxx?

challenger800, 2020-07-24:
I have the same issue before, but when i used the new bluetooth firmware IntelBluetoothFirmware.kext version 1.1.1, it seems work fine now.
My laptop is Thinkpad T470 and intel 8260.

jiaogger, 2020-07-24:
My laptop is huawei matebook 13，network card series is Intel 9560.
heliport version is 1.0.0

jiaogger, 2020-07-24:
> I will test it tonight.
> what I focus on? just only wifi signal?

yep. The wifi signal always break down while the bluetooth device is working. I checked my bluetooth options at the same time and eventually find out that there is no issue on it.

jiaogger, 2020-07-24:
> I will test it tonight.
> what I focus on? just only wifi signal?

就是用蓝牙听歌的时候wifi就没了，但耳机还是正常连接的

williambj1, 2020-07-24:
I can not reproduce this issue with

- 9560ac (A370)
- Air Pods 2
- https://github.com/OpenIntelWireless/HeliPort/commit/f621daed72560fd57d1a67ee0d7e98c57bcbaebb
- https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/5b918bf63348fc8a761ddd091a41231ce0b9732e
- https://github.com/OpenIntelWireless/itlwm/commit/9e890928f3db6fef2545eb4f7547e582a31bf3a8
- MacOS Big Sur Developer Beta 1
- MacOS Catalina 10.15.6

Kext complied with Xcode 12 and Xcode 11, with SDK macosx10.16 and macosx10.15

WiFi works perfectly when I’m listening to music.

M69W, 2020-07-24:
> I can not reproduce this issue with
> 
> * 9560ac (A370)
> * Air Pods 2
> * [OpenIntelWireless/HeliPort@f621dae](https://github.com/OpenIntelWireless/HeliPort/commit/f621daed72560fd57d1a67ee0d7e98c57bcbaebb)
> * [OpenIntelWireless/IntelBluetoothFirmware@5b918bf](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/5b918bf63348fc8a761ddd091a41231ce0b9732e)
> * itlwm [9e89092](https://github.com/OpenIntelWireless/itlwm/commit/9e890928f3db6fef2545eb4f7547e582a31bf3a8)
> * MacOS Big Sur Developer Beta 1
> * MacOS Catalina 10.15.6
> 
> Kext complied with Xcode 12 and Xcode 11, with SDK macosx10.16 and macosx10.15
> 
> WiFi works perfectly when I’m listening to music.

same 同

> I can not reproduce this issue with
> 
> * 9560ac (A370)
> * Air Pods 2
> * [OpenIntelWireless/HeliPort@f621dae](https://github.com/OpenIntelWireless/HeliPort/commit/f621daed72560fd57d1a67ee0d7e98c57bcbaebb)
> * [OpenIntelWireless/IntelBluetoothFirmware@5b918bf](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/commit/5b918bf63348fc8a761ddd091a41231ce0b9732e)
> * itlwm [9e89092](https://github.com/OpenIntelWireless/itlwm/commit/9e890928f3db6fef2545eb4f7547e582a31bf3a8)
> * MacOS Big Sur Developer Beta 1
> * MacOS Catalina 10.15.6
> 
> Kext complied with Xcode 12 and Xcode 11, with SDK macosx10.16 and macosx10.15
> 
> WiFi works perfectly when I’m listening to music.

same

- Intel AX200  
- OpenIntelWireless/HeliPort@f621dae  
- OpenIntelWireless/IntelBluetoothFirmware@5b918bf  
- OpenIntelWireless/itlwm OpenIntelWireless/itlwm@1cf4a6b

[image]

[image]

[itlwmx.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/4972385/itlwmx.kext.zip)
[HeliPort.zip](https://github.com/OpenIntelWireless/itlwm/files/4972388/HeliPort.zip)
[Bluetooth.zip](https://github.com/OpenIntelWireless/itlwm/files/4972397/Bluetooth.zip)

jiaogger, 2020-07-24:
> I can not reproduce this issue with
>
> * 9560ac (A370)
> * Air Pods 2
> * [...]

I find a possible reason for this issue, that is the frequency band the wi-fi use. 2.4ghz wi-fi might cause inter symbol interference with bluetooth. if you transform to 5ghz wi-fi, this problem can be solved automatically.

M69W, 2020-07-24:
> I find a possible reason for this issue, that is the frequency band the wi-fi use. 2.4ghz wi-fi might cause inter symbol interference with bluetooth. if you transform to 5ghz wi-fi, this problem can [...]

stable running
[image]

jiaogger, 2020-07-24:
> stable running
> 
> [image]

对，你是5Ghz的Wi-Fi，我们有问题的应该都是2.4Ghz的Wi-Fi。因为蓝牙的工作频率也在2.4Ghz左右，可能会发生串扰

jiaogger, 2020-07-24:
> stable running
> 
> [image]

你是用的heliport吗

M69W, 2020-07-24:
> > stable running
> > [image]
> 
> 你是用的heliport吗

Yes, you can see native client of the piture in right's.

jiaogger, 2020-07-25:
> Yes, you can see native client of the picture in right's.
I‘m curious at how you find out the specific information of heliport and your current situation of wifi......

M69W, 2020-07-25:
> > Yes, you can see native client of the picture in right's.
> 
> I‘m curious at how you find out the specific information of heliport and your current situation of wifi......

press and hold the option key (windows) and click the icon of the HeliPort.

jiaogger, 2020-07-26:
> > > Yes, you can see native client of the picture in right's.
> > 
> > 
> > I‘m curious at how you find out the specific information of heliport and your current situation of wifi......
> 
> press and hold the option key (windows key) and click the icon of the HeliPort.

I changed the wireless channel of my 2.4GHz wifi (initial: channel 1, now: channel 11(better change it to 6/11)) and find out that issue can be solved. I thought the issue is mainly caused by the interference of wireless channel of the 2.4GHz wifi and the bluetooth channel. Thanks for your help!

9370User, 2020-07-26:
same here with intel 8265 when ever i connect bluetooth magic mouse speed drops from 1.1mb to 50kb

jiaogger, 2020-07-26:
> > > Yes, you can see native client of the picture in right's.
> > 
> > 
> > I‘m curious at how you find out the specific information of heliport and your current situation of wifi......
> 
> press and hold the option key (windows key) and click the icon of the HeliPort.

I find a possible way to solve this issue, that is fixed the channel of 2.4GHz Wi-Fi at 13 (aka. located the work frequency as far as you can). This can commendably improve the performance of wifi. I guess the main reason of this issue is that MacOS doesn't have
 the cooperative mode between wifi and bluetooth for intel network card. Anyway, really appreciate for your help!

jiaogger, 2020-07-26:
And caution: this method CAN'T perfectly solve this issue. If you want to throughly fix this problem, please buy a router that can work in 5GHz and linked with it.

jiaogger, 2020-07-26:
There is still accidentally stuck occurs, but it's in my comfort zone.

closed 2020-08-15 by williambj1