OpenIntelWireless/itlwm #205: 9560 能连接但是无法上网？

ssdeg, 2020-07-23:
感谢Z大无私付出，我的9560使用客户端和不使用客户端都能连接，但以太网一直显示未连接，无法获取ip地址，无法正常上网，网卡：0xA370
[image]

williambj1, 2020-07-23:
https://openintelwireless.github.io/General/Troubleshooting.html#resetting-system-preferences-network

ssdeg, 2020-07-23:
> https://openintelwireless.github.io/General/Troubleshooting.html#resetting-system-preferences-network

Thanks.

ssdeg, 2020-07-23:
按上面链接操作还是未能解决问题，系统是10.15.5。
[Log_2020-07-23_21-58-47.log](https://github.com/OpenIntelWireless/itlwm/files/4966940/Log_2020-07-23_21-58-47.log)

zxystd, 2020-07-23:
> 按上面链接操作还是未能解决问题，系统是10.15.5。
> [Log_2020-07-23_21-58-47.log](https://github.com/OpenIntelWireless/itlwm/files/4966940/Log_2020-07-23_21-58-47.log)

check your WiFi password.

ssdeg, 2020-07-23:
使用

> > 按上面链接操作还是未能解决问题，系统是10.15.5。
> > [Log_2020-07-23_21-58-47.log](https://github.com/OpenIntelWireless/itlwm/files/4966940/Log_2020-07-23_21-58-47.log)
> 
> check your WiFi password.
Thanks.
使用手机wifi热点正常了。。可能是我的wifi信号不太好导致上述故障..

closed 2020-07-24 by zxystd