OpenIntelWireless/itlwm #207: Wi-Fi no longer working after wake up

igorkulman, 2020-07-24:
With the latest commit https://github.com/OpenIntelWireless/itlwm/commit/1cf4a6b6912fff6e624d280077fc336285600472 after laptop wake-up the kext is in some kind of error state (showing an exclamation mark in HeliPort).

Logs: 
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/4971715/itlwm.log)

zxystd, 2020-07-24:
no any changes in sleep/wake code, can you reproduce it or have you changed someting about efi

igorkulman, 2020-07-24:
> no any changes in sleep/wake code, can you reproduce it or have you changed someting about efi

No EFI changes, I just unloaded the kext, built it with the latest commit, loaded it again. It worked. I put the computer to sleep, woke it up later, git this problem.

I then went back to the previous commit https://github.com/OpenIntelWireless/itlwm/commit/b0f203e82ff36e79dabb691774905809cd4a8bc1, did the same and no problems after wake up from sleep.

zxystd, 2020-07-24:
try clean and then rebuild? it is a impossible problem...

igorkulman, 2020-07-24:
Ok so I tried again and it is not related to sleep and wake up.

When I just build and load the kext using https://github.com/OpenIntelWireless/itlwm/commit/1cf4a6b6912fff6e624d280077fc336285600472 the wifi does not work right away, it is in some error state.

When I build and load the kext using https://github.com/OpenIntelWireless/itlwm/commit/b0f203e82ff36e79dabb691774905809cd4a8bc1 everything works fine.

Unloading, building and loading is the same, the only difference is which commit I use.

zxystd, 2020-07-24:
sometimes the log will be corrupted after kext unloading. you can type `log show --last 1s | grep itlwm`,after do that, load the kext, and then you can see the correct log.

ErrorErrorError, 2020-07-24:
I'll also be checking and see if I can reproduce this

igorkulman, 2020-07-24:
So here is what I did

- unloaded the kext
- waited 10 minutes
- build the kext using https://github.com/OpenIntelWireless/itlwm/commit/1cf4a6b6912fff6e624d280077fc336285600472
- looked at the time to know when it will get loaded
- loaded the kext

Wifi showed as in error state (loaded 20:05 in the attached log), with a IPv6 address in HeliPort. After a short while it changed to OK state, with IPv4 address and Internet worked

So then I

- put the computer to sleep
- waited for a while
- woke up the computer

The Wifi was again in some error state, again with IPv6 address showing in HeliPort (about 20:08 in the logs). I waited but it never changed to a OK state. It switched to IPv4 address but not a DHCP assigned address but 169.254.19.29. I tried unloading the kext and loading again, still error state. F

I built https://github.com/OpenIntelWireless/itlwm/commit/b0f203e82ff36e79dabb691774905809cd4a8bc1 and still error state.

Finally I build and loaded https://github.com/OpenIntelWireless/itlwm/commit/9e890928f3db6fef2545eb4f7547e582a31bf3a8 and ater loading the kext Wifi started working just fine.

Log: 
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/4973658/itlwm.log)

igorkulman, 2020-07-24:
Might be related to https://github.com/OpenIntelWireless/itlwm/issues/160 but I have not had that issue running https://github.com/OpenIntelWireless/itlwm/commit/9e890928f3db6fef2545eb4f7547e582a31bf3a8

ErrorErrorError, 2020-07-24:
> Might be related to https://github.com/OpenIntelWireless/itlwm/issues/160 but I have not had that issue running https://github.com/OpenIntelWireless/itlwm/commit/9e890928f3db6fef2545eb4f7547e582a31bf3a8

Are you connecting to a 2.4GHz WiFi or a 5GHz WiFi? I've been having some issues with only 5GHz not connecting at all sometimes.

igorkulman, 2020-07-24:
> > Might be related to #160 but I have not had that issue running [9e89092](https://github.com/OpenIntelWireless/itlwm/commit/9e890928f3db6fef2545eb4f7547e582a31bf3a8)
> 
> Are you connecting to a 2.4GHz WiFi or a 5GHz WiFi? I've been having some issues with only 5GHz not connecting at all sometimes.

Just 2.4 Ghz

ErrorErrorError, 2020-07-24:
After trying out the same steps you did, I was unable to reproduce the issue. Mine connected fine and looking in the logs there doesn't seem to be any errors related to power state or anything related to the itlwm being off.

Intel AC9560 1st Gen 
macOS Catalina 10.15.5
Using latest itlwm commit 1cf4a6b

ErrorErrorError, 2020-07-24:
> With the latest commit [1cf4a6b](https://github.com/OpenIntelWireless/itlwm/commit/1cf4a6b6912fff6e624d280077fc336285600472) after laptop wake-up the kext is in some kind of error state (showing an exclamation mark in HeliPort).
> 
> Logs:
> [itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/4971715/itlwm.log)

The exclamation mark is due to no route issue, I have this issue only with 5GHz networks, sometimes it connects, sometimes it doesn't. It's a hit or miss situation. But as you stated you only connect to 2.4GHz, which I had issues before with that frequency range too, but rarely.

zxystd, 2020-07-25:
[image]
@ErrorErrorError can you check the WiFi password? seems that the driver receive death several times when you connect to another network, that always means the driver associate to router fail because of the wrong password,. Not really connect to router, so you can't get the IP address.

ErrorErrorError, 2020-07-25:
> [image]
> ,
> @ErrorErrorError can you check the WiFi password? seems that the driver receive deauth several times when you connect to another network, that always means the driver associate to router fail because of the wrong password,. Not really connect to router, so you can't get the IP address.

Ah, It was because of that, wtf, Atleast now I see why it sometimes fails, thanks lol

evertjr, 2020-07-25:
After the latest changes the driver no longer connect to my 5ghz network too, only the 2.4ghz. I reverted to 2020-07-22 and it worked fine (from here https://github.com/1hbb/OpenIntelWireless-Factory/releases)

zxystd, 2020-07-25:
> After the latest changes the driver no longer connect to my 5ghz network too, only the 2.4ghz. I reverted to 2020-07-22 and it worked fine (from here https://github.com/1hbb/OpenIntelWireless-Factory/releases)

any logs？

evertjr, 2020-07-25:
> > After the latest changes the driver no longer connect to my 5ghz network too, only the 2.4ghz. I reverted to 2020-07-22 and it worked fine (from here https://github.com/1hbb/OpenIntelWireless-Factory/releases)
> 
> any logs？

How can I record usefull logs to you?

zxystd, 2020-07-25:
@evertjr `log show --last boot | grep itlwm >~/Desktop/itlwm.log`, then you can see a log file on the Desktop.

zxystd, 2020-07-25:
@igorkulman I will try it on my test machine soon. now I am using ax200 on my laptop, works fine.

evertjr, 2020-07-25:
@zxystd see if you can spot something wrong. The itlwm2-4.log is the working connection on 2.4ghz, the other is trying on 5ghz. 
[itlwm2-4.log](https://github.com/OpenIntelWireless/itlwm/files/4974870/itlwm2-4.log)
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/4974871/itlwm.log)

zxystd, 2020-07-25:
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/4975217/itlwm.kext.zip)
@igorkulman @evertjr revert some changes, can you help me to test?

igorkulman, 2020-07-25:
> [itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/4975217/itlwm.kext.zip)
> @igorkulman @evertjr revert some changes, can you help me to test?

This build seems to be working fine, when I load it it connects to my network defined in Info.plist almost right away and Internet works fine, even after sleep.

evertjr, 2020-07-25:
@zxystd I can confirm this fixed the problem!

zxystd, 2020-07-25:
@igorkulman @evertjr @ErrorErrorError ok, thank you.

closed 2020-07-26 by igorkulman