OpenIntelWireless/itlwm #209: Leaks while unloading kext

ErrorErrorError, 2020-07-25:
Hey,

Every time I try to unload the kext I can never seem to be able to unload it.
Error: 
``` 
Build/Products/Debug
➜ sudo kextunload itlwm.kext                                          22:52:04
(kernel) Can't unload kext com.zxystd.itlwm; classes have instances:
(kernel)     Kext com.zxystd.itlwm class itlwm_interface has 1 instance.
Failed to unload com.zxystd.itlwm - (libkern/kext) kext is in use or retained (cannot unload).
```

Using the latest kext commit: 1cf4a6b6912fff6e624d280077fc336285600472

zxystd, 2020-07-25:
I can unload with itlwm/itlwmx, how do you do that?

ErrorErrorError, 2020-07-25:
Well, after experimenting looks like I partially fixed the issue. Loading itlwm from a bootloader, i.e. OpenCore, causes the kext to not be able to unload sometimes. Same thing with loading the kext after the system is up and running. I am not sure why it sometimes unloads and sometimes it does not but as of now it does let me unload if I load the kext manually rather than the bootloader loading the kext.

ErrorErrorError, 2020-07-26:
sorry, accidentally closed case

Goshin, 2020-07-26:
> EDIT: I noticed it does not unload if I open HeliPort, even if I close it before unloading kext.

So the problem is in Heliport. Currently I don't have an intel network adapter to test with. Maybe @ErrorErrorError can help locate the function or the line to blame. I guess the leak might happen in `bool open_adapter(io_connect_t *connection_t)`, in `api.c`.

ErrorErrorError, 2020-07-26:
> So the problem is in Heliport. Currently I don't have an intel network adapter to test with. Maybe @ErrorErrorError can help locate the function or the line to blame. I guess the leak might happen [...]

Thats what I suspected at first but I didn't find any leaks in that method or anywhere else in api.c. And right now I am able to unload the kext sometimes with heliport running.

Goshin, 2020-07-26:
Heliport creates a one-time connection to itlwm for every API call to ensure itlwm is unloadable when Heliport is running. The problem might be caused by HeliPort exiting with an unfinished API call in the background. I made a fix in https://github.com/OpenIntelWireless/HeliPort/commit/ae5f199e42c4189c18a6f756866fcda5a9881c1d, hope it can solve the issue.

ErrorErrorError, 2020-07-27:
> Heliport creates a one-time connection to itlwm for every API call to ensure itlwm is unloadable when Heliport is running. The problem might be caused by HeliPort exiting with an unfinished API call in the background. I made a fix in [OpenIntelWireless/HeliPort@ae5f199](https://github.com/OpenIntelWireless/HeliPort/commit/ae5f199e42c4189c18a6f756866fcda5a9881c1d), hope it can solve the issue.

Awesome! Looks like it fixed the issue in my end! Thanks for the hard work @Goshin. 👍

zxystd, 2020-07-29:
@ErrorErrorError Can confirm it would happened? if so, please close the issue. tks.

closed 2020-07-29 by ErrorErrorError