OpenIntelWireless/itlwm #211: Basic tutorial for itlwm [EN]

assanges, 2020-07-25:
This is a translation of #146 w/ limited or no support at all.

## [Original article on the PC Beta forum](http://bbs.pcbeta.com/forum.php?mod=viewthread&tid=1848662) (Simp. Chinese only)
### _This tutorial is based on the driver and app presented/ complied on 2020-06-18, which may differ significantly from the current release. Please stay tuned for updates!_
## Prequestive: Driver and software
### Driver - kext
Complie the [itlwm S/C](https://github.com/OpenIntelWireless/itlwm)

Or download `itlwmx_ax200_0618.kext.zip` via PC Beta above 
(Note: only the itlwmx kext w/ an `x` can be applied on [Intel AX200](https://ark.intel.com/content/www/cn/zh/ark/products/189347/intel-wi-fi-6-ax200-gig.html))

### Application - HeliPort
Compile the [HeliPort S/C](https://github.com/OpenIntelWireless/HeliPort)

#### The following driver & app have been compiled for you: (Source:`黑苹果Intel WiFi交流群`)
[HeliPort.zip](https://github.com/OpenIntelWireless/itlwm/files/4808904/HeliPort.zip)（as in JUN18）
[itlwmx.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/4808905/itlwmx.kext.zip)（as inJUN18）
 
## Apply the magic
If you are not going to use the `HeliPort` app,
Please remember to modify the `SSID`和 `password` in the `itlwmx.kext`/`Info.plist`
i.e.:【`itlwmx.kext`->RMB->Show Package Contents】:`Contents/Info.plist`
[image]
_Oh BTW, some of you might encounter insufficiecnt rights after editing the `info.plist`, and solution beneath_
 
According to the author, @zxystd, stated on the PC Beta forum (the link above), the kext driver needs to be loaded manually w/ Terminal.
 
```
If you want to add new Wifi connection & don't want to reboot your device, 
you can unload the kext -> modify it -> re-compile it -> reload it
Unload: 
sudo kextunload -b com.zxystd.itlwm
Load:
sudo chown -R root:wheel itlwm.kext
sudo kextutil -v 6 itlwm.kext
```

Okie Dokie, will need to clarify the steps above for dummies, the kext path might differ.
So let's take `AX200` (`itlwmx.kext`) for demo:
First, extract the `itlwmx.kext` to `Desktop` (i.e. the path of driver=Desktop);
Open terminal and type `sudo chown -R root:wheel `
then drag the `itlwmx.kext` into the terminal, yep, you look it right, drag it inside.
After that you'll get a `sudo chown -R root:wheel /Users/abc/Desktop/itlwmx.kext`
instead `sudo chown -R root:wheel itlwm.kext`, you know what I mean.

Normally, if you not going to put the kext into your OC or Clover partition,
you would only need two lines to execute in the Terminal in order to load it up:

```
sudo chown -R root:wheel itlwmx.kext
sudo kextutil -v 6 itlwmx.kext
```
After you run these 2 lines👆,
Open 【 Launchpad -> Sys. Pref. -> Network -> (Bttm-left corner) ->【+ 】-> choose the【Ethernet】-> Apply
[image]

The open the `HeliPort`，
You can try to click the `Wi-Fi` icon multiple times if you didn't see the list (just like moving the cursor in WIN95 for I/O lol)
As in the current timestamp (`2020-06-18`), HeliPort Support joining Wi-Fi connection, other functions unknown due to lack of test. The application shall be improved in later stages, stay tuned.

## Troubleshoot
1. How do I know if I have connected to Wi-Fi？
   Open 'HeliPort',【Open Net Pref. -> Network -> check Status: Connected (Green light)】
2. What if I don't have any【Ethernet】option to add:
Use [小新pro13 EFI OC](https://github.com/daliansky/XiaoXinPro-13-hackintosh) as example
Comment w/ `#` or remove 2 entries: (I think `ASPM` is very suspicious，Details [PCI Device ASPM](https://github.com/daliansky/OC-little/tree/master/01-%E5%85%B3%E4%BA%8EAOAC/01-5-%E8%AE%BE%E7%BD%AEASPM%E5%B7%A5%E4%BD%9C%E6%A8%A1%E5%BC%8F), but unconfirmed）

```
DeviceProperties\Add\PciRoot(0x0)/Pci(0x1C,0x0)
DeviceProperties\Add\PciRoot(0x0)/PciRoot(0x0)/Pci(0x1C,0x0)/Pci(0x0,0x0)
```

[image]

3. How do I quit `HeliPort` ：
Credit to @williambj1：

```
a) Alt (Cmd) + LMB on the HeliPort, you shall pass through;
b) sudo killall HeliPort  in the Terminal ;
```

or c) the most complicated way ( maybe somebody just can't help themself being geek, huh? )
Open 【 Lauchpad -> Other -> Activity Monitor -> Find【 HeliPort】, double clicks ->【Quit】

4. Wi-Fi Connections after Wake from sleep：
Theoretically, it shall reconnect after it woke up,
howeeeeevvverrrr if that's not the case:
w/ HeliPort : 1. Turn off Wi-Fi; 2. Turn on Wi-Fi, if you've configured a default SSID to, it should do its job again.
 
### Optimization/ Automization
Put the `itlwmx.kext` into `OC` or `Clover` partition, it will do its job, no more Terminal typing.
Moreover, if the `Info.plist` has been configured w/ a`Wi-Fi SSID` & `password`, even no Heliport is required. 

### DON'T ASK THINGS LIKE HOW TO AIRDROP, it's in the early stage mate!
### This tutorial is 'as-in' in 2020-06-18, which may differ significantly from the current release. Please stay tuned for updates!
### Finally a big thank-you to every contributor for their efforts on intel wifi being able to use on macOS, thanks again to @zxystd
### (And my most appreciation for @M69W for writing tutorial for dummies, thanks!)

assanges, 2020-07-25:
大佬們請收下我的膝蓋 <(_ _)>

williambj1, 2020-07-25:
会整合进教程，感谢

M69W, 2020-07-25:
非常感谢！赞

Stevemoretz, 2020-07-27:
Thanks for this but isn't it a little simpler than this?I just put Helliport in application folder and kext loaded the kext and that was it.

M69W, 2020-07-27:
It's simple but no easy for some one.

assanges, 2020-07-30:
> Thanks for this but isn't it a little simpler than this?I just put Helliport in application folder and kext loaded the kext and that was it.

Your comment described this in the TL;DR fashion perfectly LUL

closed 2020-09-19 by williambj1