OpenIntelWireless/itlwm #214: ac9462/ax201/ac9560 2th gen Intel Card Supporting

zxystd, 2020-07-27:
I had tried about one week to investigate how to make it work on macOS, but still found nothing. The real problem is that the card can cause interrupt on macOS, I don't know why, I think the firmware upload codes are pretty well, but make no sense. listed problem.
1. is gen2 cards only support MSI-X interrupt on macOS, if so, how to use MSI-X on macOS?
2. is there something should cooperate with cards work, because CNVIO2 cards are special?

BimLee, 2020-07-27:
I think the following link will help you
https://developer.apple.com/forums/thread/26966

zxystd, 2020-07-27:
@BimLee Thank you for your information, but still confused, the problem is same with mine in some way, but no solutions including there.

mrkaratechamp, 2020-07-27:
i'm very instersting in help, I have a 2gen card in my matebook 13 2020, so I can test whatever you need

leoli-dev, 2020-07-28:
I have the problem with my Gigabyte z490i mobo, if there is anything I can do for helping please let me know.

agassecond, 2020-07-28:
Not sure about MacOS, but looks like it uses same interupt model as ssd in Ubuntu
https://linux-hardware.org/index.php?probe=915ab6a317&log=interrupts

djdjz7, 2020-07-28:
I have the 2nd gen 9560 WiFi card in my laptop, please let me know if I can help testing.

serpinu, 2020-07-29:
hi!, i have the ax201 if i can help i'm here! :)

Jaydar, 2020-07-31:
i have the  i7-10710U + ax201

Xoloth, 2020-08-01:
i have AX201 (but windows shows its AC9462).

purecmd, 2020-08-01:
Kernel Developer's Manual(OpenBSD)
https://man.openbsd.org/pci_intr_map.9

lulujyc, 2020-08-01:
I have AX201 although its PCI ID should be AC9462. Its id is VEN_8086 and DEV_02F0 and shows itself as "Intel Wi-Fi 6 AX201 160Mhz" after installed driver under windows. I'm using 10510U with OpenCore and Big Sur. Current itlwmx driver doesn't support the card. I can help to test if possible.

zxystd, 2020-08-01:
[itlwmx_0802_beta1.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5011153/itlwmx_0802_beta1.kext.zip)
Now it is supported, try it.

yuhr-zz, 2020-08-01:
听到 ac9560 2nd 能用了，非常开心。感谢您的无私付出！

我用最新的源代码编译后测试的，发现还是无法使用。

我的机器是华为 Matebook 13（i7十代CPU 2020款 ）有以下的网卡：
Intel AC9560 (9462)
VendorID 0x8086 DeviceID 0x02F0 SubVendorID 0x8086 SubDeviceID 0x2034

以下是测试的Log等。

**kextstat | grep -i itlwmx**

get the following result:

   _74    0 0xffffff7f85173000 0x3aa000   0x3aa000   com.zxystd.itlwmx (1.0.0) DACC3C3A-8B34-3C98-A1EC-A132C57B4EA2 <18 13 6 5 3 1>_

**log show --last boot | grep itlwmx**

get the following result:

_2020-08-02 06:54:53.696579+0800 0xe6       Default     0x0                  0      0    kernel: (itlwmx) _start(struct kmod_info*, void*) has been invoked
2020-08-02 06:54:53.696707+0800 0xe6       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.696758+0800 0xe8       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.696760+0800 0xe7       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.697553+0800 0xec       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.697578+0800 0xea       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.697602+0800 0xeb       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.697624+0800 0xed       Default     0x0                  0      0    kernel: (itlwmx) itlwm: probe msi capa exists
2020-08-02 06:54:53.697653+0800 0xed       Default     0x0                  0      0    kernel: (itlwmx) itlwm: probe msix capa exists
2020-08-02 06:54:53.697974+0800 0xee       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.697996+0800 0xf9       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.698030+0800 0xf6       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.698221+0800 0xf5       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.698912+0800 0xfa       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.699429+0800 0xf2       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.699429+0800 0xf8       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.699442+0800 0xf0       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.699743+0800 0xf7       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.699754+0800 0xe9       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.702508+0800 0xf4       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.702533+0800 0xf3       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.702974+0800 0xf1       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:53.702976+0800 0xef       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:56.407017+0800 0x54b      Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:56.408558+0800 0x552      Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 06:54:56.419135+0800 0x549      Default     0x0                  0      0    kernel: (itlwmx) itlwm: free_

yuhr-zz, 2020-08-02:
手动 load 的log

Mac-MB13 !_AppleIntelWIFI_3 % sudo -S kextutil -v 6 ./itlwmx.kext
Password:
Kext user-space log filter changed from 0xff2 to 0xfff.
Kext kernel-space log filter changed from 0xff2 to 0xfff.
Kext library architecture set to x86_64.
Can't stat /AppleInternal - No such file or directory.
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext library recording diagnostics for: validation authentication dependencies warnings.
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Kext with invalid signature (-67030) allowed: <OSKext 0x7fa8fdd055e0 [0x7fff85a43cc0]> { URL = "file:///Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext/", ID = "com.zxystd.itlwmx" }
Code Signing Failure: code signature is invalid
/Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext appears to be loadable (not including linkage for on-disk libraries).
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Loading /Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext.
Reading load info for 9 kexts.
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
"com.apple.kpi.bsd" = {
    "OSBundleWiredSize" = 50016
    "CFBundleIdentifier" = "com.apple.kpi.bsd"
    "OSBundleUUID" = <655aa406 97024364 9d304815 71cf631d>
    "OSBundleCompatibleVersion" = "8.0.0b1"
    "CFBundleVersion" = "19.6.0"
    "OSBundleStarted" = true
    "OSKernelResource" = true
    "OSBundleLoadTag" = 1
    "OSBundleLoadAddress" = -551890460672
    "OSBundlePath" =
        "/System/Library/Extensions/System.kext/PlugIns/BSDKernel.kext"
    "OSBundleRetainCount" = 115
    "OSBundlePrelinked" = true
    "OSBundleIsInterface" = true
    "OSBundleLoadSize" = 50016
}
"com.apple.kpi.libkern" = {
    "OSBundleWiredSize" = 65712
    "CFBundleIdentifier" = "com.apple.kpi.libkern"
    "OSBundleUUID" = <95e6d3e2 13da4030 9ea28de9 337472d5>
    "OSBundleCompatibleVersion" = "8.0.0d0"
    "CFBundleVersion" = "19.6.0"
    "OSBundleStarted" = true
    "OSKernelResource" = true
[... 171 lines cut ...]
(kernel) Loading kext com.zxystd.itlwmx.
(kernel) Kext com.apple.kpi.bsd is already loaded.
(kernel) Kext com.apple.kpi.mach is already loaded.
(kernel) Kext com.apple.iokit.IONetworkingFamily is already loaded.
(kernel) Kext com.apple.iokit.IONetworkingFamily sending 2 personalities to the IOCatalogue and starting matching.
(kernel) Kext com.apple.kpi.iokit is already loaded.
(kernel) Kext com.apple.kpi.libkern is already loaded.
(kernel) Kext com.apple.iokit.IOPCIFamily is already loaded.
(kernel) Kext com.apple.iokit.IOPCIFamily sending 3 personalities to the IOCatalogue and starting matching.
(kernel) Allocated link buffer for kext com.zxystd.itlwmx at 0xffffff7f8549e000 (4063232 bytes).
(kernel) Kext com.zxystd.itlwmx executable loaded; 992 pages at 0xffffff7f8529e000 (load tag 171).
(kernel) Kext com.zxystd.itlwmx calling module start function.
(kernel) Kext com.zxystd.itlwmx registered class itlwmx_interface.
(kernel) Kext com.zxystd.itlwmx has IOService subclass itlwmx_interface; enabling autounload.
(kernel) Kext com.zxystd.itlwmx registered class ItlNetworkUserClient.
(kernel) Kext com.zxystd.itlwmx registered class IOTaskQueue.
(kernel) Kext com.zxystd.itlwmx registered class pci_intr_handle.
(kernel) Kext com.zxystd.itlwmx registered class itlwmx.
(kernel) Kext com.zxystd.itlwmx registered class CTimeout.
(kernel) Kext com.zxystd.itlwmx is now started.
(kernel) Kext com.zxystd.itlwmx sending 1 personality to the IOCatalogue and starting matching.
(kernel) Kext com.zxystd.itlwmx loaded.
Successfully loaded /Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext.
/Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext successfully loaded (or already loaded).
Invalid signature -67030 for kext <OSKext 0x7fa8fdd055e0 [0x7fff85a43cc0]> { URL = "file:///Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext/", ID = "com.zxystd.itlwmx" }

yuhr-zz, 2020-08-02:
Mac-MB13 !_AppleIntelWIFI_3 % log show --last boot | grep itlwmx
2020-08-02 08:23:42.666285+0800 0x30ac     Default     0x0                  1151   0    sudo:     yuhr : TTY=ttys000 ; PWD=/Users/yuhr/!_AppleIntelWIFI_3 ; USER=root ; COMMAND=/usr/bin/kextutil -v 6 ./itlwmx.kext
2020-08-02 08:23:42.968119+0800 0x30c8     Default     0x0                  0      0    kernel: (itlwmx) _start(struct kmod_info*, void*) has been invoked
2020-08-02 08:23:42.970796+0800 0x30d4     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.971480+0800 0x30da     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.971643+0800 0x30e4     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972167+0800 0x30de     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972214+0800 0x30e8     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972379+0800 0x30e3     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972549+0800 0x30d7     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972640+0800 0x30d6     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972825+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: probe msi capa exists
2020-08-02 08:23:42.972857+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: probe msix capa exists
2020-08-02 08:23:42.972983+0800 0x30d5     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.973243+0800 0x30d8     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.973446+0800 0x30e0     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.973649+0800 0x30e5     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.984214+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_prepare_card_hw
2020-08-02 08:23:42.984217+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-08-02 08:23:43.060478+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: : Failed to wake up the nic
2020-08-02 08:23:43.060484+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_ifdetach
2020-08-02 08:23:43.060486+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_proto_detach
2020-08-02 08:23:43.060488+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_crypto_detach
2020-08-02 08:23:43.060490+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_node_detach
2020-08-02 08:23:43.060492+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ifmedia_delete_instance
2020-08-02 08:23:43.060614+0800 0x74       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:24:43.062503+0800 0x16d8     Default     0x0                  0      0    kernel: (itlwmx) _stop(struct kmod_info*, void*) has been invoked

yuhr-zz, 2020-08-02:
Mac-MB13 !_AppleIntelWIFI_3 % kextstat | grep -i itlwmx
  172    0 0xffffff7f8529e000 0x3aa000   0x3aa000   com.zxystd.itlwmx (1.0.0) DACC3C3A-8B34-3C98-A1EC-A132C57B4EA2 <18 13 6 5 3 1>

yuhr-zz, 2020-08-02:
Mac-MB13 !_AppleIntelWIFI_3 % sudo -S kextutil -v 6 ./itlwmx.kext
Password:
Kext user-space log filter changed from 0xff2 to 0xfff.
Kext kernel-space log filter changed from 0xff2 to 0xfff.
Kext library architecture set to x86_64.
Can't stat /AppleInternal - No such file or directory.
Defaulting to kernel file '/System/Library/Kernels/kernel'
Kext library recording diagnostics for: validation authentication dependencies warnings.
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Kext with invalid signature (-67030) allowed: <OSKext 0x7fe4ac4136a0 [0x7fff85a43cc0]> { URL = "file:///Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext/", ID = "com.zxystd.itlwmx" }
Code Signing Failure: code signature is invalid
/Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext appears to be loadable (not including linkage for on-disk libraries).
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
}

Loading /Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext.
Reading load info for 9 kexts.
Reading loaded kext info from kernel.
(kernel) User-space log flags changed from 0x0 to 0xfff.
(kernel) Received 'Get Loaded Kext Info' request from user space.
(kernel) Returning loaded kext info.
Loaded kext info:
{
"com.apple.kpi.bsd" = {
    "OSBundleWiredSize" = 50016
    "CFBundleIdentifier" = "com.apple.kpi.bsd"
    "OSBundleUUID" = <655aa406 97024364 9d304815 71cf631d>
    "OSBundleCompatibleVersion" = "8.0.0b1"
    "CFBundleVersion" = "19.6.0"
    "OSBundleStarted" = true
    "OSKernelResource" = true
    "OSBundleLoadTag" = 1
    "OSBundleLoadAddress" = -551890460672
    "OSBundlePath" =
        "/System/Library/Extensions/System.kext/PlugIns/BSDKernel.kext"
    "OSBundleRetainCount" = 115
    "OSBundlePrelinked" = true
    "OSBundleIsInterface" = true
    "OSBundleLoadSize" = 50016
}
"com.apple.kpi.libkern" = {
    "OSBundleWiredSize" = 65712
    "CFBundleIdentifier" = "com.apple.kpi.libkern"
    "OSBundleUUID" = <95e6d3e2 13da4030 9ea28de9 337472d5>
    "OSBundleCompatibleVersion" = "8.0.0d0"
    "CFBundleVersion" = "19.6.0"
    "OSBundleStarted" = true
    "OSKernelResource" = true
    "OSBundleLoadTag" = 5
    "OSBundleLoadAddress" = -551890530304
[... 171 lines cut ...]
(kernel) Kext com.apple.iokit.IONetworkingFamily is already loaded.
(kernel) Kext com.apple.iokit.IONetworkingFamily sending 2 personalities to the IOCatalogue and starting matching.
(kernel) Kext com.apple.kpi.iokit is already loaded.
(kernel) Kext com.apple.kpi.libkern is already loaded.
(kernel) Kext com.apple.iokit.IOPCIFamily is already loaded.
(kernel) Kext com.apple.iokit.IOPCIFamily sending 3 personalities to the IOCatalogue and starting matching.
(kernel) Allocated link buffer for kext com.zxystd.itlwmx at 0xffffff7f8549e000 (4063232 bytes).
(kernel) Kext com.zxystd.itlwmx executable loaded; 992 pages at 0xffffff7f8529e000 (load tag 172).
(kernel) Kext com.zxystd.itlwmx calling module start function.
(kernel) Kext com.zxystd.itlwmx registered class itlwmx_interface.
(kernel) Kext com.zxystd.itlwmx has IOService subclass itlwmx_interface; enabling autounload.
(kernel) Kext com.zxystd.itlwmx registered class ItlNetworkUserClient.
(kernel) Kext com.zxystd.itlwmx registered class IOTaskQueue.
(kernel) Kext com.zxystd.itlwmx registered class pci_intr_handle.
(kernel) Kext com.zxystd.itlwmx registered class itlwmx.
(kernel) Kext com.zxystd.itlwmx registered class CTimeout.
(kernel) Kext com.zxystd.itlwmx is now started.
(kernel) Kext com.zxystd.itlwmx sending 1 personality to the IOCatalogue and starting matching.
(kernel) Kext com.zxystd.itlwmx loaded.
Successfully loaded /Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext.
/Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext successfully loaded (or already loaded).
Invalid signature -67030 for kext <OSKext 0x7fe4ac4136a0 [0x7fff85a43cc0]> { URL = "file:///Users/yuhr/!_AppleIntelWIFI_3/itlwmx.kext/", ID = "com.zxystd.itlwmx" }

Mac-MB13 !_AppleIntelWIFI_3 % kextstat | grep -i itlwmx
  172    0 0xffffff7f8529e000 0x3aa000   0x3aa000   com.zxystd.itlwmx (1.0.0) DACC3C3A-8B34-3C98-A1EC-A132C57B4EA2 <18 13 6 5 3 1>

yuhr-zz, 2020-08-02:
Mac-MB13 !_AppleIntelWIFI_3 % kextstat | grep -i itlwmx
  172    0 0xffffff7f8529e000 0x3aa000   0x3aa000   com.zxystd.itlwmx (1.0.0) DACC3C3A-8B34-3C98-A1EC-A132C57B4EA2 <18 13 6 5 3 1>
yuhr@Mac-MB13 !_AppleIntelWIFI_3 % log show --last boot | grep itlwmx
2020-08-02 08:23:42.666285+0800 0x30ac     Default     0x0                  1151   0    sudo:     yuhr : TTY=ttys000 ; PWD=/Users/yuhr/!_AppleIntelWIFI_3 ; USER=root ; COMMAND=/usr/bin/kextutil -v 6 ./itlwmx.kext
2020-08-02 08:23:42.968119+0800 0x30c8     Default     0x0                  0      0    kernel: (itlwmx) _start(struct kmod_info*, void*) has been invoked
2020-08-02 08:23:42.970796+0800 0x30d4     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.971480+0800 0x30da     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.971643+0800 0x30e4     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972167+0800 0x30de     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972214+0800 0x30e8     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972379+0800 0x30e3     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972549+0800 0x30d7     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972640+0800 0x30d6     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.972825+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: probe msi capa exists
2020-08-02 08:23:42.972857+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: probe msix capa exists
2020-08-02 08:23:42.972983+0800 0x30d5     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.973243+0800 0x30d8     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.973446+0800 0x30e0     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.973649+0800 0x30e5     Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:23:42.984214+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_prepare_card_hw
2020-08-02 08:23:42.984217+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_set_hw_ready
2020-08-02 08:23:43.060478+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: : Failed to wake up the nic
2020-08-02 08:23:43.060484+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_ifdetach
2020-08-02 08:23:43.060486+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_proto_detach
2020-08-02 08:23:43.060488+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_crypto_detach
2020-08-02 08:23:43.060490+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_node_detach
2020-08-02 08:23:43.060492+0800 0x30db     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ifmedia_delete_instance
2020-08-02 08:23:43.060614+0800 0x74       Default     0x0                  0      0    kernel: (itlwmx) itlwm: free
2020-08-02 08:24:43.062503+0800 0x16d8     Default     0x0                  0      0    kernel: (itlwmx) _stop(struct kmod_info*, void*) has been invoked
2020-08-02 08:30:41.246314+0800 0x4677     Default     0x0                  1372   0    sudo:     yuhr : TTY=ttys000 ; PWD=/Users/yuhr/!_AppleIntelWIFI_3 ; USER=root ; COMMAND=/usr/bin/kextutil -v 6 ./itlwmx.kext

moonheart, 2020-08-02:
感谢！现在可以正常使用了 小新pro13 2020 i7-10710u

lulujyc, 2020-08-02:
谢谢，AX201现在可以使用2.4g网络，但是5g网络连接不太稳定

lulujyc, 2020-08-02:
使用这个版本的itlwmx还会导致启动时间延长5-10秒，不知道是不是个例😂

Syedayy, 2020-08-02:
On my Huawei Matebook D15 with Intel AC9560/9462
(VendorID 0x8086 DeviceID 0x02F0 SubVendorID 0x8086 SubDeviceID 0x2034), itlwmx does not work and after running `log show --last boot | grep itlwmx`, I get the same results as yuhr-zz.

Heliport also says "itlwm is not running" 🤔

zxystd, 2020-08-02:
@Syedayy Try the latest pre-release.

Syedayy, 2020-08-02:
It works now! 
[image] here's my speeds on my 5 Ghz WiFi. My usual download/upload speed is 100/20.

ErrorErrorError, 2020-08-02:
> It works now!
> [image] here's my speeds on my 5 Ghz WiFi. My usual download/upload speed is 100/20.

It's as expected in the current state of itlwm/x. In the future the speeds will improve!

id86021, 2020-08-02:
感谢您的无私付出！0x06F0  AX201可以使用了。

mrkaratechamp, 2020-08-02:
amazing, you´re a genius!
works on my Matebook 13 2020, itlwmx  driver, the .zip here it's not working, but your latest commit, works perfect, just download and compile with Xcode.

the speed is still not al 100%, but it´s totally works for now,
thanks!!!

yuhr-zz, 2020-08-02:
@zxystd 感谢你的付出！现在完美使用中！

purecmd, 2020-08-02:
Thank you so much.
[image]

nekiro, 2020-08-02:
Could someone upload latest kext for testing?

easyworld, 2020-08-02:
Awesome! It works on Huawei Matebook X Pro 2020
[image]

supdo, 2020-08-02:
我的matebook14，2020款，ac9560可是上网了。

leoli-dev, 2020-08-02:
> [itlwmx_0802_beta1.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5011153/itlwmx_0802_beta1.kext.zip)
> Now it is supported, try it.

This kext works correctly on my Hackintosh, my mobo is using AX201. Thanks a million!

I changed the pwd/ssid in the kext and put in into EFI, it connect immediately to my wifi:
[image]

Here's the speed test result with itlwmx:
[image]

Here's the speed test result with a USB wifi adapter:
[image]

moosetk, 2020-08-04:
> 感谢您的无私付出！0x06F0 AX201可以使用了。

您好，我的也是ax201，我要怎么样让这个驱动生效呢？运行 sudo kextutil -v 6 itlwmx.kext 的时候总是 Untrusted kexts are not allowed

williambj1, 2020-08-04:
关闭 sip 或者用引导注入

Issue solved, closed.

closed 2020-08-04 by williambj1

david34corbalan, 2022-03-02:
hello i have the same problem i have a dell latitude 5510 laptop with ax201, and i can't get the wifi to appear