OpenIntelWireless/itlwm #215: itlwm is not running

songmings, 2020-07-27:
All versions have this problem，but when me use command "kextstat |grep itlwm"，can saw the itlwm is running but in haliport,this app is display :itlwm is not running,in network of operating system,could find an network is Ethernet,but it was could not got IP address,so i have reinstalled the operating system many times but it is can not be worked

ErrorErrorError, 2020-07-28:
What intel card do you have?

songmings, 2020-07-28:
> What intel card do you have?

AC 8265，i have checked the wifi adapter，it is not working ,it is broken,i don't know why,Will this driver cause the network card to break down？

Skimige, 2020-07-28:
Me too here, using AX200, self-built version of https://github.com/OpenIntelWireless/itlwm/commit/2b407cb87b5d96ea8600d3ead49f5fa373d635d7 . itlwmx does load and run, but not connected to WiFi, HeliPort also reports itlwm is not running.

closed 2020-07-28 by songmings