OpenIntelWireless/itlwm #216: I'm facing a weird issue with my wifi on hackintosh

swaroop2208, 2020-07-28:
My wifi in fluctuating continuesly even though I updated to new version can any give me suggestions how to fix it.
It would be really helpful for me.

zxystd, 2020-07-28:
check the log by the command.
`log show --last boot | grep itlwm >~/Desktop/itlwm.log`
and send it to me.

swaroop2208, 2020-07-28:
> check the log by the command.
> `log show --last boot | grep itlwm >~/Desktop/itlwm.log`
> and send it to me.
[image]

zxystd, 2020-07-28:
get the log file from your desktop, and upload here.

swaroop2208, 2020-07-28:
itlwm.log

zxystd, 2020-07-28:
log file, not file name, drag the file here and wait for upload done, then commit your comment.

swaroop2208, 2020-07-28:
[system.log](https://github.com/OpenIntelWireless/itlwm/files/4989346/system.log)
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/4989347/itlwm.log)

zxystd, 2020-07-29:
the driver is worked, but your wifi password is wrong, check it, BTW, the driver version you are using is too old, download the latest. thanks.

swaroop2208, 2020-07-29:
for your info wifi worked for 2 hours once i restarted it stopped working.
can u give me latest link for drivers.

Syedayy, 2020-07-29:
https://github.com/OpenIntelWireless/itlwm/releases

itlwm under assets.

swaroop2208, 2020-07-29:
thank you will try...

swaroop2208, 2020-08-02:
My issue have been solved as I updated my intel wi-fi and Bluetooth kexts and updates oc snapshot thanks for helping.

closed 2020-08-02 by swaroop2208