OpenIntelWireless/itlwm #220: Call for contributors on reddit

notrexbias, 2020-07-31:
Hey guys, 
@usr-sse2 , @lgariv , @Skimige , @zxystd , @ErrorErrorError 
I've been using this project for around a month now and I'm loving it. So, I reached out to reddit for assistance, it'd be nice and helpful if any of you can join the conversation over there and answer some questions that I cannot since there are people who really want to assist.
Here's the link to the discussion: [Reddit](https://www.reddit.com/r/hackintosh/comments/i19l77/call_for_contributors_for_an_open_source_intel/)

ErrorErrorError, 2020-07-31:
I can help out on Reddit. There is also a guide on how to install https://openintelwireless.github.io/ however I would recommend non-developers to use the alpha version, of course there will be bugs but if people are curious to try it out.

notrexbias, 2020-07-31:
Yes please, assist whenever you can because I may not have all the answers for the questions that are being asked on that thread.
I'm a programmer but I've never delved into creating kexts and so this is the best way I  thought I would contribute and also thanks for replying

md-siam, 2020-08-01:
If you want to include "Bangla Translation" then I want to contribute to this project.

zxystd, 2020-08-02:
We have Gitter, you can go to here if you need help.
https://gitter.im/OpenIntelWireless/itlwm
BTW, I can't go to Reddit because it is blocked by China DNS FireWall.

notrexbias, 2020-08-02:
@zxystd thanks

closed 2020-08-15 by williambj1