OpenIntelWireless/itlwm #221: This is not a issue, but a query

Parthvsquare, 2020-07-31:
[image]
i came from reddit call for contributor. 
using LG Gram 14 (8265dw2)
i was using the hard wired version. i made a automator app and was not able to use sip for a long time. thanks guys for making this possible.
i have no issue with sleep, it takes a sec or few to restart. it's just and hour well keep updating

Parthvsquare, 2020-07-31:
i haven't used the (firmware) itlmx.kext. just popped itlm to C/O, removed the files as requested from library->Preferences->systemConfiguration

ErrorErrorError, 2020-07-31:
> this is how it should turn out right?

Yes that is supposed to turn out. itlwm/x creates an ethernet interface since Apple80211, macOS's Wi-Fi, is closed source.

 

> i haven't used the (firmware) itlmx.kext. just popped itlm to C/O, removed the files as requested from library->Preferences->systemConfiguration and reboot that's it

You should only use itlwmx if you have a gen 2 intel device. If you don't then you should be using itlwm.

williambj1, 2020-08-01:
https://openintelwireless.github.io/

Please read our docs

closed 2020-08-01 by williambj1