OpenIntelWireless/itlwm #225: NUC8 Intel 9560 Wifi Problems

Ibrahim778, 2020-08-01:
For some reason my wifi only works when I run it on the SECOND Reboot. For example after a cold start it doesnt work reboot and its fine. Same for windows I have to go into macOS once and reboot again to make it work otherwise Heliport doesn't show any wifi networks. I'm on Catalina btw.

williambj1, 2020-08-01:
1. Please specify your kext and heliport version
2. Provide logs for failures and successes
3. Test again with a Linux live USB

Based on your info, it's impossible to identify whether it's your issue or ours.

zxystd, 2020-08-01:
Try remove All the Intel kexts, and try again if it is still happened in Windows.
And, fill all necessary info like @williambj1  said, so we can locate where is the problem. thanks.

Ibrahim778, 2020-08-01:
Right. Sorry It's getting late now and I'm Busy tomorrow so I'll do what you said the day after. I never had these problems in windows or Linux. Thanks you very much for your quick response!  :)

zxystd, 2020-08-01:
OK, waiting for your further response, if problem exists, maybe we need your help to locate the problem and test, thanks.

Ibrahim778, 2020-08-04:
OK back now so how do I provide log files? I don't actually know the version because I found this kext and heliport on a subreddit a while ago but I think It's fairly recent.

ErrorErrorError, 2020-08-04:
> OK back now so how do I provide log files? I don't actually know the version because I found this kext and heliport on a subreddit a while ago but I think It's fairly recent.

Please use the latest version, I.e. the latest compiled. You can get it here https://github.com/OpenIntelWireless/itlwm/releases/

Ibrahim778, 2020-08-05:
Ok. Ill report back with the new version soon

williambj1, 2020-08-15:
Please reopen if necessary.

closed 2020-08-15 by williambj1