OpenIntelWireless/itlwm #226: WiFi disconnects

Parthvsquare, 2020-08-01:
## Platform:
- LG gram 14 with 8265d2w and using latest 1.1.1 intelbluetooth kext
- have aptx headphones connected
- wifi range is 3-4 bars jumps in between
--- 
## How to recreate:
- there is not problem when listening to spotify 
- also when surfing through internet no issue
- Youtube and other video platform wifi disconnects

zxystd, 2020-08-01:
try to use 5g WiFi, thanks.

Parthvsquare, 2020-08-02:
Sorry i don't have 5gz router

zxystd, 2020-08-02:
Bluetooth and 2.4ghz Wifi work in same channel band, they will influence with each other certainly. If you don't have 5ghz WiFi router, the only way is to find and change the WiFi channel(eg: channel 13) and lock to it, maybe will more better.

Parthvsquare, 2020-08-02:
Thanks mate. But I already tried this, still was having random disconnect

chris1111, 2020-08-11:
I having also random disconnect with ➤ [PCE-AC55BT-B1](https://www.asus.com/ca-en/Networking/PCE-AC55BT-B1/)
I have also freeze loading Safari page I need to Disable Wifi on menue bar then enable and the qWifi reconnect

javiernferraro, 2020-08-17:
Hi! I´m having the same issue. I´m on a Gigabyte z490 vision d that has a cnvi intel 6 card.
Everything works ok, but suddenly I get randomly disconnected, I just click on Heliport´s icon and it works again.
Any help would be appreciated. Maybe if I try using https://github.com/usr-sse2/Black80211-Catalina this?

chris1111, 2020-08-17:
> Hi! I´m having the same issue. I´m on a Gigabyte z490 vision d that has a cnvi intel 6 card.
> Everything works ok, but suddenly I get randomly disconnected, I just click on Heliport´s icon and it [...]

I have already try then it produce the same random disconnect

zxystd, 2020-08-17:
@chris1111 @javiernferraro can't help you without running logs, you can get it thought heliport, press option key and left click the wifi icon, click the diagnosis and upload the log and tell me the time the problem happened.

javiernferraro, 2020-08-17:
Hi! I just got one at 20:16... 
[bugreport_61126.zip](https://github.com/OpenIntelWireless/itlwm/files/5087258/bugreport_61126.zip)

zxystd, 2020-08-18:
@javiernferraro your router inform to disconnect, maybe you are doing heavy upload foreground or background? or you can switch a router, be more near from router and test again, thanks.
[image]