OpenIntelWireless/itlwm #227: HeliPort not working

Babyvox550, 2020-08-02:
[image]
[image]
[image]
Heliport still not working!!!!

OC as follows:

[image]
[image]

What should I do next?

Babyvox550, 2020-08-02:
[image]
Hackintool showed wifi is drived.

1Revenger1, 2020-08-02:
Get logs and see if itlwm attached right. The PCIe section of Hackintool doesn't show if the driver attached, Hackintool has it's own way to recognize PCIe devices.

https://openintelwireless.github.io/General/Troubleshooting.html

zxystd, 2020-08-02:
Use the latest pre-release kext will help you fix this. Ac9462 certainly can work now.

closed 2020-08-07 by zxystd