OpenIntelWireless/itlwm #229: Battery draining issue

swaroop2208, 2020-08-02:
Hey everyone I'm using acer nitro 5 hackintosh for a week now and I've observed that my battery backup has reduced to half of its usual performance.
Need help to fix it.

zxystd, 2020-08-03:
Would you mind doing a comparative experiment? Compare the rate of discharge when playing video or doing other things with and without Wi Fi, and want to know the data.

swaroop2208, 2020-08-03:
> Would you mind doing a comparative experiment? Compare the rate of discharge when playing video or doing other things with and without Wi Fi, and want to know the data.

Ya I've done that thing like playing video on 1080p in youtube I just got 2 hours backup..
And haven't tired without WiFi cause I have good wifi connection at my home and.

dvoina, 2020-08-04:
The same happens on Huawei Matebook 13 2020. 
The kernel_task eats between 58% and 80% of CPU. 

[https://imgur.com/a/mPeDcHX](url)

Driver is correctly loaded:
root@Daniels-MBP ~ # kextstat | grep itlwmx
   72    0 0xffffff7f83922000 0x7f9000   0x7f9000   com.zxystd.itlwmx (1.0.0) 1E6BD343-ED02-37D4-8412-AFC4566C2648 <29 13 6 5 3 1>

closed 2020-08-09 by swaroop2208