OpenIntelWireless/itlwm #230: will it be supported EDUROAM for University Campus

mehmetyukcell, 2020-08-03:
I am an PhD student in the university. will it be supported for eduroam internet connection?

ErrorErrorError, 2020-08-03:
As of now there are still things that need to be implemented before you will be able to connect to networks that require Username and password

mehmetyukcell, 2020-08-03:
> As of now there are still things that need to be implemented before you will be able to connect to networks that require Username and password 

what are these things? I have username and password. i will try to connect next week with Eduroam in the campus.

ErrorErrorError, 2020-08-03:
https://github.com/OpenIntelWireless/HeliPort/issues/4

mehmetyukcell, 2020-08-03:
> https://github.com/OpenIntelWireless/HeliPort/issues/4
> 

Thanks for your interest, Sincerely

williambj1, 2020-08-15:
Duplicate of #101

closed 2020-08-15 by williambj1