OpenIntelWireless/itlwm #232: itlwm works after a few restarts

FaneCH, 2020-08-03:
When i try to load the driver, it shows in terminal that it loaded successful but nothing happens (heliport shows itlwm not running). After a few restarts, it connects.
I'm running macOS Catalina 10.15.6 and my WiFi card is Intel AC 3165.

[image]

ErrorErrorError, 2020-08-03:
Please upload logs https://openintelwireless.github.io/General/Troubleshooting.html#kernel-extensions

FaneCH, 2020-08-03:
Got it to work after a few reboot and deleting the stuff from network.

[image] 

There are the logs:
[Log_2020-08-03_19-49-54.log](https://github.com/OpenIntelWireless/itlwm/files/5017201/Log_2020-08-03_19-49-54.log)

Silfalion, 2020-08-05:
> Got it to work after a few reboot and deleting the stuff from network.
> 
> [image]
> 
> There are the logs:
> [Log_2020-08-03_19-49-54.log](https://github.com/OpenIntelWireless/itlwm/files/5017201/Log_2020-08-03_19-49-54.log)

Could you specify the things you deleted please?

FaneCH, 2020-08-05:
In the network pane (sys pref) i had two connections: one for bluetooth, one for ethernet (from a usb wifi dongle).
I deleted those 2 things and after a couple of tries, it worked. This is my problem, itlwm is not loading everytime after a reboot even if the output of kextload/ kexutil says so.

Silfalion, 2020-08-05:
Thank for the details. Encountered that problem as well. I restarted twice and it works now but only tested for 2 restarts.

zxystd, 2020-08-07:
@FaneCH @Silfalion I think that's yours EFI issue, the driver loaded, but can't find the network device, driver is passive load and match by system, if the system didn't pass the PCI device, how can the driver work?

Silfalion, 2020-08-07:
I’ll retest now with a few restarts and btw, thank you so much. I’m over the top to have wifi working at mostly full speed right now.

Silfalion, 2020-08-07:
just tried now, on cold boot it doesn't work, the driver loads, heliports launches and detects the wifi, connects to it but always have an exclamation mark on it. After a restart it works(writing these lines from the Hackintosh). Any ideas?

zxystd, 2020-08-07:
@Silfalion if the Heliport can show the network, that means the driver is working, if the network didn't get a IP address then Heliport will show an exclamation mark on it. for more detail, you should give me to log. 
`log show --last boot | grep itlwm >~/Desktop/itlwm.log`
also, tell me the time the problem happened.

Silfalion, 2020-08-07:
Here is the log file, on cold start, same problem:

[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5040851/itlwm.log)

zxystd, 2020-08-07:
@Silfalion everything works fine, the network is connected, just wait for the ip address. or you can change the router channel 1 to others.

Silfalion, 2020-08-07:
Thing is, it connects immediately after restarting, whether I wait 10 mn on a cold start or 1. On cold start it never connects despite how long I wait.

FaneCH, 2020-08-08:
@zxystd i putted itlwm.kext into kexts folder (OC) and after log in, it doesn't connect to internet. Only timewhen it connects to internet is when i load it manually from terminal.
[Log_2020-08-08_15-48-04.log](https://github.com/OpenIntelWireless/itlwm/files/5045266/Log_2020-08-08_15-48-04.log)

zxystd, 2020-08-24:
@FaneCH edit your OC config.plist, configure the itlwm.kext injection info.

zxystd, 2020-08-24:
close. this is a EFI issue.

closed 2020-08-24 by zxystd