OpenIntelWireless/itlwm #233: [Question] merge with Black80211-Catalina?

igorkulman, 2020-08-04:
I found this project https://github.com/usr-sse2/Black80211-Catalina that combines https://github.com/AppleIntelWifi/Black80211-Catalina and itlwm so itlwm looks like a Wi-Fi card not an Ethernet adapter to the system and Heliport is not needed.

It does not work with itlwm as is, but it requires using a fork https://github.com/usr-sse2/itlwm.

Any thoughts on merging Black80211-Catalina to itlwm directly so it looks like a Wi-Fi card?

1Revenger1, 2020-08-04:
Black80211-Catalina requires the headers to be modified every time that the 80211 headers change within macOS, which is basically every major release of macOS. Not only that, it's a pain to develop with as you can't unload/load it at will without retention errors iirc.

Itlwm afaik presents itself as an ethernet controller to make it easier to develop and to make it so that the headers don't need to be reversed every time IO80211Controller changes.

vit9696, 2020-08-05:
In short, we plan to upstream this, but it is a bit too early for a pull request. Kext loading/unloading on the fly is removed in macOS 11.0, so this is not a problem anyway. As for compatibility, the changes are abstracted well enough to support both front-ends chosen at compile-time: HeliPort and Black80211, so I do not see a problem here as well.

CC @usr-sse2

Stevemoretz, 2020-09-17:
So what is it too painful to have slightly different versions for different Macos versions?Or maybe this could even been solved in the code I don't know what headers are, that you're talking about.

williambj1, 2020-09-17:
> Or maybe this could even been solved in the code

Impossible

Stevemoretz, 2020-09-17:
Alright I'm not sure what it is, but I've taken it right, it's a slightly different thing for each masos release, and it can't be fixed in the code, so how about using some automation tool to change the portion of code build it change it again build it and so on till it's over. I use gulp so to stuff like this that I can't solve in the code.I use gulp to solve it outside the code.Using a bit of regex you can change the code easy and...

ErrorErrorError, 2020-09-17:
> Alright I'm not sure what it is, but if I've taken it right, it's a slightly different thing for each masos release, and it can't be fixed in the code, so how about using some automation tool to change the portion of code build it change it again build it and so on till it's over. I use gulp to do stuff like this that I can't solve in the code.I use gulp to solve it outside the code.Using a bit of regex you can change the code easily and...
> 
> 
> 
> I could help with regex and stuff if you think this way could work.

In theory that would sound awesome, but unfortunately it's not possible to "automate" what change and what has not win C.

1Revenger1, 2020-09-17:
itlwm already uses C preprocessing to import a different set of headers depending on what version you specify you want to build for. So it's pretty easy to change through the build settings.

Stevemoretz, 2020-09-17:
Build settings are probably in a text file right?

And still could use gulp to change that text file if you needed, it's totally possible.Gulp is an automation tool it doesn't care what language or what, it can change all the text files.Run commands and...

usr-sse2, 2020-09-18:
As @zxystd decided to make his own reimplementation of the same, merging Black80211-Catalina is not necessary, and I won't continue working on it. I'll switch my efforts to making Apple RSNSupplicant work with AirportItlwm, which I already achieved in Black80211-Catalina and which enables WPA2 Enterprise (802.1X EAP). @zxystd, is it acceptable or you would race again and implement the same?

zxystd, 2020-09-18:
@usr-sse2 @vit9696 I think you should have misunderstood the meaning of airportitwm. These days, I have been thinking about how to work with black80211 in a better way. Finally, I decided to merge itlwm and itlwmx, and create a new airport driver. You can understand it as an implementation method of airport code template. Most of the code of airportitwm was in my private branch a few months ago, so I can easily to merge it, and added my research and implementation related to airport in recent days. Your's black80211 is really a great project, especially the scanning part and WPA2 The implementation of enterprise encryption, and I believe there will be more amazing function implementation. I am very happy to merge them and cooperate to develop more functions. our ultimate goal is to make the Intel wifi driver more perfect. I don't think that these works can be completed by one or two people. This is my original intention of open source. I recommend that we can use gitter to communicate instantly. Of course, we can also give code push permission to this repo if necessary.

vit9696, 2020-09-18:
@zxystd yeah, I guess the lack of communication sorted of caused some double work, because if we knew you had private code and wanted to integrate it earlier (for example, lastly @usr-sse2 spent a lot of time investigating WPA2 Enterprise and resolving sleep issues), we could have prioritised the work differently.

Anyway, I think we could live with gitter. Both @usr-sse2 and I are in your support room, so it will suit if you create one more for the discussion. I think currently besides the minor cleanup of the code and conflict resolving there are three things to discuss:

1. Merging of RSN supplicant to AirportItlwm, as from what I understood from @usr-sse2 using AirportItlwm as a base may be easier.
2. Deciding on the architecture of the solution. Black80211 was designed to let different drivers (if anybody wants to port them) to implement IO80211Family support rather seamlessly. AirportItlwm makes it more Intel-centric, which may be reasonable given the amount of efforts needed to implement Wi-Fi stack without Apple support, but we are still undecided.
3. We are planning to make a new repository with Mac kernel SDK to contain all currently missing headers in Xcode, reverse-engineered headers, and headers adopted with availability macros to allow targeting cleanly for older macOS versions. We would want IO80211Family headers to live there for everyone convenience.

mishurov, 2020-10-02:
@vit9696 my guess: reversing headers consists of reading symbols in binaries and demangling the names.

williambj1, 2020-10-16:
Done!

closed 2020-10-16 by williambj1

LeaveNhA, 2020-10-20:
Wow.