OpenIntelWireless/itlwm #235: Network dont respond

naveenkrdy, 2020-08-05:
Hi !
i observed a weird issue.
my laptop (with intel wifi) is connected to the same wifi network to which my other hackintosh is connected too through ethernet. So whenever i boot my other hackintosh that is connected through ethernet. my internet on the laptop stops working for some reason. To fix it i have to switch off and switch on my laptop wifi.

log stream output:
```
log stream | fgrep itlwm                                                                                               ✘130 10:49:41
2020-08-05 10:54:53.172421+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_eapol_key_input
2020-08-05 10:54:53.172454+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_recv_rsn_group_msg1
2020-08-05 10:54:53.172463+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_eapol_key_mic
2020-08-05 10:54:53.172756+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_must_update_group_key
2020-08-05 10:54:53.172767+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_set_link_state nstate=4, old_state=4
2020-08-05 10:54:53.172782+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: : received msg 1/2 of the group key handshake from c4:e9:0a:5c:b2:ac
2020-08-05 10:54:53.172794+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_send_group_msg2
2020-08-05 10:54:53.172809+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: : sending msg 2/2 of the group key handshake to c4:e9:0a:5c:b2:ac
2020-08-05 10:54:53.172820+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_send_eapol_key
2020-08-05 10:54:53.172828+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_eapol_key_mic
2020-08-05 10:55:13.958908+0530 0x2428     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_free_node
2020-08-05 10:55:13.958928+0530 0x2428     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_free_node
2020-08-05 10:55:13.958938+0530 0x2428     Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_free_node
2020-08-05 10:55:13.958947+0530 0x2428     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_bgscan
2020-08-05 10:55:13.958949+0530 0x2428     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_umac_scan
2020-08-05 10:55:13.958958+0530 0x2428     Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_send_cmd: sending command 0x10d
2020-08-05 10:55:13.961044+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: iwx_cmd_done: command 0x10d done
2020-08-05 10:55:14.058753+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_setup_node c4:e9:0a:5c:b2:ac
2020-08-05 10:55:14.132262+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_setup_node 90:8d:78:75:fb:b0
2020-08-05 10:55:24.657990+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: : end background scan
2020-08-05 10:55:24.658022+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_node_choose_bss
2020-08-05 10:55:24.658046+0530 0x15f      Default     0x0                  0      0    kernel: (itlwmx) itlwm: ieee80211_reset_scan
```

[image]

[bugreport_40962.zip](https://github.com/OpenIntelWireless/itlwm/files/5026319/bugreport_40962.zip)

Regards !

zxystd, 2020-08-07:
The log didn't show any exceptions, everything works fine. maybe it is a router compatible issue? I have wired and wireless devices at home, they all works fine, can you try another router? thanks.

williambj1, 2020-09-12:
Closed due to no further response.

closed 2020-09-12 by williambj1