OpenIntelWireless/itlwm #236: Warm Reboot Issue

DellG3, 2020-08-06:
Hello Developers!
I have an AC9462 and the warm reboot issue still remains in the new build of itlwm. There's no network connection when restarted from Windows to Mac OS. Even restarting Mac OS at this point doesn't work either. It works only when Mac OS is the first that I boot after shut down.
Furthermore, there's slight unresponsiveness when surfing the internet. Pages take more time to load. (Confirmed not the ISP issue)
What are more things to be fixed till we get a release? Believe in Intel!

zxystd, 2020-08-06:
which kext do you used?
what's your card productId?
any logs? `log show --last boot | grep itlwm >~/Desktop/itlwm.log`

DellG3, 2020-08-06:
Hackintool reported that I had AC9560. Sorry about that.
Product id: 0xA370
I used itlwm compiled on August 2.
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5037345/itlwm.log)
Thanks!

DellG3, 2020-08-06:
Attaching another log when successfully running on First boot from shut down.
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5037460/itlwm.log)

zxystd, 2020-08-07:
according to the first log, the PCI device didn't match the driver(network card dismiss?), so it is your EFI issue. thanks.

DellG3, 2020-08-08:
I'm unable to find any EFI issues. itlwm works fine every time when cold boot. Is there anything I can do as a temporary fix?
Like loading the kext manually from the desktop? It's strange too that Windows shows that I have an AC9462.

zxystd, 2020-08-09:
@DellG3 Have you inject PCI info about ASPM value? My friend have same problem with you, but after replace with my EFI, everything goes fine.

DellG3, 2020-08-12:
Sorry, I didn't understand. I have only put the kext to Open Core EFI and did no further inject. Kindly elaborate what I should do.
This can possibly fix the problem and help me understand more.

metro2450, 2020-08-24:
> Sorry, I didn't understand. I have only put the kext to Open Core EFI and did no further inject. Kindly elaborate what I should do.
> This can possibly fix the problem and help me understand more.

I can confirm I use to have this problem on my AX200 card yesterday when I installed the alpha kext.

Today I update to the stable released kext and everything works fine. Now after reboot I can still use WiFi.  Therefore I suggest you update to the latest release version.

DellG3, 2020-08-30:
@metro2450 , the new stable kext didn't fix my problem. Dear @zxystd, kindly elaborate how your friend fixed this issue.

zxystd, 2020-09-03:
@DellG3 Don't know too much about Hackintosh so can't help you, but my friend fix this problem when replace his whole EFI. 
https://github.com/daliansky/XiaoXinPro-13-hackintosh/releases
this is the EFI, for your reference.

closed 2020-09-19 by williambj1

metro2450, 2020-09-23:
> Sorry, I didn't understand. I have only put the kext to Open Core EFI and did no further inject. Kindly elaborate what I should do.
> This can possibly fix the problem and help me understand more.

I have faced the same problem and observed the solution as injecting in opencore to disable ASPM
Under your WiFi card PCI device tree, add the following 
Key = pci-aspm-default | Value = 0 | Type = NUMBER