OpenIntelWireless/itlwm #238: Does not connect to my main wifi, but does connect to my hotspot

dytractial, 2020-08-07:
whenever i use heliport to input the password for my home wifi, it doesn't connect and all of the wifi hotspots it found would go away and then come back. it does connect to my phones hotspot though

i tried doing the same in linux, and it both connected. my home wifi uses wpa/wpa2 personal but my phone uses wpa2

dytractial, 2020-08-07:
also im using an intel wireless 3160

dytractial, 2020-08-07:
i switched to my killer ax 1650 (just a rebranded ax200) and i get the exact same problem only this time when i try to connect to my home network the other networks dont dissappear.

ErrorErrorError, 2020-08-07:
Upload logs when trying to connect to your home network. Wait about 2 mins after you tried to connect to your home network. You can create log diagnostics by clicking command + WiFi icon and clicking "Create Diagnostics"

dytractial, 2020-08-07:
i did, and all the log said was "Failed to connect to Oldham WiFi"

when looking through console, i get the exact same thing. not descriptive at all

ErrorErrorError, 2020-08-07:
> when looking through console, i get the exact same thing. not descriptive at all

Please upload logs as requested

dytractial, 2020-08-08:
@
[itlwmx_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5044349/itlwmx_logs.log)
[HeliPort_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5044350/HeliPort_logs.log)

dytractial, 2020-08-10:
Still need help, currently stuck with using usb tethering on my android phone.

ErrorErrorError, 2020-08-10:
> Still need help, currently stuck with using usb tethering on my android phone.

Hm, reading the logs looks like it does re initializes, as it should when connecting to a network, but it then just scans again once it finishes loading. Okay can you install intel wireless 3160 and use itlwm.kext and upload logs after trying to connect?

dytractial, 2020-08-10:
I will once i get the chance, the antenna cables are very tedious to connect

zxystd, 2020-08-10:
@dytractial try to write your said and password in driver's info.plist, close Heliport and restart.

dytractial, 2020-08-12:
i get the same error when i do, i think it has to do with my wifi being wpa/wpa2

teewhydope, 2020-08-14:
> @dytractial try to write your ssid and password in driver's info.plist, close Heliport disable Heliport auto start and restart machine.

please can you add support for 9560 AC with IDs   0x8086:0x9DB1 and  0x8086:0x9DB1

i'm unable to connect to any hotspot with heliport

dytractial, 2020-08-27:
im still at a loss with this, my last resort to getting it to work is to login to my router and change the encryption style from wpa/wpa2 (automatic selecting) to just wpa

im guessing the reason why it's not connecting is maybe the driver isn't letting the router know which encryption standard to use? i also connected to my neighbours (i asked them) which uses wpa, and it connected just fine.

ErrorErrorError, 2020-08-28:
What is your router model?

dytractial, 2020-08-28:
mine's a ZyXel VMG4825-B10A

codeVerine, 2020-09-04:
I also have the exact same issue. Heliport is not connecting to my home WiFi, but freely connecting to my iPhone hotspot.

**What I have tried**

- Changed my WiFi encryption to WPA2 AES same as iPhone hotspot. Yet not working.
- Set iPhone hotspot password to home network as well, no luck.
- Removed home wifi password, still not connecting.
- I did not try adding said and password to info.plist since @dytractial already tried it without any luck.

This bug is really pestering me as otherwise this is a wonderful project. My router is not dual band so I don't have a 5ghz WiFi to test. I can help you by any means to fix this issue.

dytractial, 2020-09-09:
my iphone uses wpa aes, not wpa2. maybe it has something to do with wpa2, but my router autodetects if the device is using wpa or wpa2, so i have no idea why it's doing this.

festano92, 2020-09-11:
Hi, I have the same issue as well with 8265 :/
It has worked for a few days perfectly with my home wifi, then nothing. Heliport and Network settings show "Connected" but the pages keep loading without actually show anything.

codeVerine, 2020-09-12:
@dytractial My iPhone hotspot shows as WPA2 AES on heliport saved networks section. I don't think it's a password encryption related issue. As I said in my previous comment, I have made my home WiFi open and tried connecting to it. Still it was not connecting. 

@zxystd @ErrorErrorError I can get you logs or anything to investigate this issue in details. It looks like more people are reporting this issue. I checked the trouble shooting page, it says to get log to check whether the itlwm is loading correctly or not. In my case I believe it's loading properly since I can connect to other hotspots without any issue. Is there any other logs that I can get for you ? Diagnostic logs are attached. Heliport shows the signal waves animation even though the heliport logs says failed to connect to network.
[HeliPort_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5212585/HeliPort_logs.log)
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5212586/itlwm_logs.log)

@festano92 My issue is a bit different. In my case, the heliport is not connecting to home WiFi. I can see the SSID listed and click on it to connect. The icon will keep showing the signal wave animation, but won't connect.

zxystd, 2020-09-12:
@dytractial @codeVerine Have you try the 1.1.0 alpha version? If problem persist, I will upload a kext contains verbose logs that you can try the get the whole logs.

codeVerine, 2020-09-12:
@zxystd Just tried it out after you asked. Still the same issue. It's not connecting to my home WiFi. Logs are attached.
[HeliPort_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5212677/HeliPort_logs.log)
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5212678/itlwm_logs.log)

It would be great if you could share the debug kext.

williambj1, 2020-09-12:
https://gitter.im/OpenIntelWireless/itlwm?at=5f5cc4783651184d30e70d34

closed 2020-09-12 by williambj1

matte-oss, 2024-09-07:
> https://gitter.im/OpenIntelWireless/itlwm?at=5f5cc4783651184d30e70d34

Hey, Sorry for necroing, but what did that link Say, i can't open It.

I have the same exact issue