OpenIntelWireless/itlwm #242: Cannot load kext issue

richardkindom, 2020-08-10:
I am using 10.13.5 and tried to kextload -v itlwm.kext, bit failed to work.(link error)
After using kextlibs I found:

Richards-MBP:tmp richard$ kextlibs itlwm.kext/
For all architectures:
    com.apple.iokit.IONetworkingFamily = 3.4
    com.apple.iokit.IOPCIFamily = 2.9
    com.apple.kpi.bsd = 17.6
    com.apple.kpi.iokit = 17.6
    com.apple.kpi.libkern = 17.6
    com.apple.kpi.mach = 17.6

For x86_64:
    14 symbols not found in any library kext.

Richards-MBP:tmp richard$ kextlibs -undef-symbols itlwm.kext/
For all architectures:
    com.apple.iokit.IONetworkingFamily = 3.4
    com.apple.iokit.IOPCIFamily = 2.9
    com.apple.kpi.bsd = 17.6
    com.apple.kpi.iokit = 17.6
    com.apple.kpi.libkern = 17.6
    com.apple.kpi.mach = 17.6

For x86_64:
    14 symbols not found in any library kext:
	__ZN19IONetworkController29_RESERVEDIONetworkController2Ev
	__ZN8OSObject8DispatchE5IORPC
	__ZN18IONetworkInterface28_RESERVEDIONetworkInterface9Ev
	__ZN12IOUserClient8DispatchE5IORPC
	__ZN19IONetworkController29_RESERVEDIONetworkController5Ev
	__ZN18IONetworkInterface28_RESERVEDIONetworkInterface5Ev
	__ZN19IONetworkController29_RESERVEDIONetworkController4Ev
	__ZN18IONetworkInterface28_RESERVEDIONetworkInterface6Ev
	__ZN15OSMetaClassBase8DispatchE5IORPC
	__ZN19IONetworkController29_RESERVEDIONetworkController3Ev
	__ZN18IONetworkInterface28_RESERVEDIONetworkInterface7Ev
	__ZN18IONetworkInterface29_RESERVEDIONetworkInterface10Ev
	__ZN9IOService8DispatchE5IORPC
	__ZN18IONetworkInterface28_RESERVEDIONetworkInterface8Ev

Do I need installing new libs?

williambj1, 2020-08-10:
Where did you download this kext? Please re-download it from GitHub Release since it's built with SDK macosx10.12,

Builds shared from other sources usually have your issue.

williambj1, 2020-08-12:
Closed due to inactivity

closed 2020-08-12 by williambj1