OpenIntelWireless/itlwm #243: Audio distortion in big sur beta

swaroop2208, 2020-08-10:
Recently updated to big sur from Catalina.
What I've found is when playing video the audio from speaker are fine.
But when it comes to headphones I'm getting disturbance I've checked few more headphones too same issue.
Need help.

ErrorErrorError, 2020-08-10:
> But when it comes to headphones I'm getting disturbance I've checked few more headphones too same issue.

Are headphones running from bluetooth? If not then that is another issue.

swaroop2208, 2020-08-10:
> > But when it comes to headphones I'm getting disturbance I've checked few more headphones too same issue.
> 
> Are headphones running from bluetooth? If not then that is another outside of itlwm/x.

Nope they are wired headphones.
So do I need to add one more itlwn/x file to my kexts.

williambj1, 2020-08-10:
@swaroop2208 I don't see any relationship between itlwm and your issue, could you please specify?