OpenIntelWireless/itlwm #245: No concertion to Magic Mousse

chris1111, 2020-08-11:
I'm using ➤ [PCE-AC55BT-B1](https://www.asus.com/ca-en/Networking/PCE-AC55BT-B1/)
No device detect ?
[image]

ErrorErrorError, 2020-08-11:
This is the wrong repository for this issue. It should be in IntelBluetoothFirmware. Also, run `kextstat | grep IntelBluetoothFirmware` in terminal to check if it's loaded.

chris1111, 2020-08-11:
> This is the wrong repository for this issue. It should be in IntelBluetoothFirmware. Also, run `kextstat | grep IntelBluetoothFirmware` in terminal to check if it's loaded.
I closed the issue I disconnect the BT cable and put in the Box
Thanks

closed 2020-08-11 by chris1111