OpenIntelWireless/itlwm #246: 无法连接转接型wifi

panguangwei6, 2020-08-11:
8.2版本目前各功能都比较完善了，在这里十分的感谢各位开发者，但使用过程中，发现了一个bug，例如：同学的笔记本开热点无法连接，会出现接上几秒又断开的反复场景，家里的转接wifi也是出现的同样的场景（客厅的wifi可以连上，但转接客厅wifi的路由器用8.2连接不上）

williambj1, 2020-08-11:
请提供日志

panguangwei6, 2020-08-11:
我在终端输入这个代码但是日志为空
[image]

ErrorErrorError, 2020-08-11:
> 我在终端输入这个代码但是日志为空
> [image]

使用高级菜单打开HeliPort，然后单击“创建诊断报告”。上载zip文件。

Edit: Sorry I don't speak Chinese however I hope I can help out.

panguangwei6, 2020-08-11:
这是我房间wifi转接客厅wifi出现的问题，连房间wifi出现的问题，因为放假回家了，没有另外一台笔记本开热点无法连接的日志，十分抱歉
[bugreport_19153.zip](https://github.com/OpenIntelWireless/itlwm/files/5054557/bugreport_19153.zip)

panguangwei6, 2020-08-11:
> > 我在终端输入这个代码但是日志为空
> > [image]
> 
> 使用高级菜单打开HeliPort，然后依次“创建诊断报告”。上载zip文件。
> 
> 编辑：对不起，我不会说中文，但是希望我能帮上忙。

谢谢！

panguangwei6, 2020-08-11:
> 请提供日志

日志已发

williambj1, 2020-08-11:
> [bugreport_19153.zip](https://github.com/OpenIntelWireless/itlwm/files/5054557/bugreport_19153.zip)
@zxystd

zxystd, 2020-08-11:
mtu调到最低1280，然后再试。
网络-》高级-〉硬件-》mtu

panguangwei6, 2020-08-11:
> mtu调到最低1280，然后再试。
> 网络-》高级-〉硬件-》 mtu

已调1280尝试，还是无法连接
[image]
[bugreport_63103.zip](https://github.com/OpenIntelWireless/itlwm/files/5054746/bugreport_63103.zip)

zxystd, 2020-08-11:
那没办法了，从日志来看一切正常，但是发过去的所有包都被路由拒绝了，估计你那个分享是p2p分享的Wi-Fi，这个是不支持的，换个路由器吧。

closed 2020-08-15 by zxystd