OpenIntelWireless/itlwm #251: AC3160 not connecting to Wi-Fi

mastercreeper66, 2020-08-12:
Hi!
I have an AC3160, i'm using the itlwm kext inside the OC EFI partition. When i try to connect to my wi-fi nothing happens. i tried rebooting multiple times and i also resetted the network preferences.
Here's the log:
[Log_2020-08-12_22-31-21.log](https://github.com/OpenIntelWireless/itlwm/files/5065588/Log_2020-08-12_22-31-21.log)

ErrorErrorError, 2020-08-12:
> Hi!
> I have an AC3160, i'm using the itlwm kext inside the OC EFI partition. When i try to connect to my wi-fi nothing happens. i tried rebooting multiple times and i also resetted the network [...]

There is a lot of authentications so make sure you are typing your password correctly.

mastercreeper66, 2020-08-12:
> 
> There is a lot of authentications so make sure you are typing your password correctly.

The password is correct

ErrorErrorError, 2020-08-12:
```
2020-08-12 22:30:30.580183+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_disassoc
2020-08-12 22:30:30.583661+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: ASSOC -> SCAN
2020-08-12 22:30:43.645089+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_assoc
2020-08-12 22:30:43.645268+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: AUTH -> ASSOC
2020-08-12 22:30:43.645305+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: sending assoc_req to 60:14:66:fc:18:77 on channel 4 mode 11g type=0
2020-08-12 22:30:43.654017+0200 0x124      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: association failed (status 1) for 60:14:66:fc:18:77
2020-08-12 22:30:48.594134+0200 0x161      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: association timed out for 60:14:66:fc:18:77
2020-08-12 22:30:48.594177+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_disassoc
2020-08-12 22:30:48.599040+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: ASSOC -> SCAN
2020-08-12 22:31:01.660695+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_assoc
2020-08-12 22:31:01.660868+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: AUTH -> ASSOC
2020-08-12 22:31:01.660891+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: sending assoc_req to 60:14:66:fc:18:77 on channel 4 mode 11g type=0
2020-08-12 22:31:01.666962+0200 0x124      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: association failed (status 1) for 60:14:66:fc:18:77
2020-08-12 22:31:06.609444+0200 0x161      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: association timed out for 60:14:66:fc:18:77
2020-08-12 22:31:06.609475+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_disassoc
2020-08-12 22:31:06.614460+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: ASSOC -> SCAN
2020-08-12 22:31:19.673398+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: iwm_assoc
2020-08-12 22:31:19.673539+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: AUTH -> ASSOC
2020-08-12 22:31:19.673553+0200 0x149      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: sending assoc_req to 60:14:66:fc:18:77 on channel 4 mode 11g type=0
2020-08-12 22:31:19.679635+0200 0x124      Default     0x0                  0      0    kernel: (itlwm) itlwm: itlwm: association failed (status 1) for 60:14:66:fc:18:77
```

According to the logs the password seemed to be incorrect. Try to connect again and using heliPort, in the advanced options click "Create Diagnostics Report..." and upload the zip file.

mastercreeper66, 2020-08-12:
[bugreport_21171.zip](https://github.com/OpenIntelWireless/itlwm/files/5066204/bugreport_21171.zip)

closed 2020-08-13 by mastercreeper66