OpenIntelWireless/itlwm #256: WiFi disconnects after a few seconds

redouglas39, 2020-08-14:
I have the ac7265 WiFi card and the WiFi disconnects itself from a few seconds to a minute after i turn on WiFi on the heliport. The WiFi fluctuates and I don’t have the same problem when using windows. Here’s the log. Thanks!
[log.txt](https://github.com/OpenIntelWireless/itlwm/files/5075162/log.txt)

alessiomatricardi, 2020-08-14:
I've also this card (pci 0x095a) but i haven't this problem. It seems works well

williambj1, 2020-08-14:
Try adjusting the MTU value in sys prefs

redouglas39, 2020-08-14:
> Try adjusting the MTU value in sys prefs

I have adjusted to the minimum (1200) and maximum (1500) and same behavior is still occurred

zxystd, 2020-08-14:
seems driver is trying to reconnect to ap after background scan, you are too far away from router? i will take a look tomorrow for the detail.

redouglas39, 2020-08-15:
> seems driver is trying to reconnect to ap after background scan, you are too far away from router? i will take a look tomorrow for the detail.

Oh yes the connection seems better when I moved it near the router. Yet the signal I got is still weaker than in windows 10. Thanks!

closed 2020-08-15 by redouglas39