OpenIntelWireless/itlwm #257: Cannot enable WiFi after sleep

mastercreeper66, 2020-08-14:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.0.0 alpha
- WiFi Card Model: AC3160
- Product ID: 0x08B4
- macOS Version: 10.14.6

**Description**

When the computer wakes from sleep, sometimes the wifi gets disabled and cannot be enabled again until i reboot

**Bug Report Archive**

[bugreport_28378.zip](https://github.com/OpenIntelWireless/itlwm/files/5077136/bugreport_28378.zip)

**Upload your Kext**

[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5077142/itlwm.kext.zip)

ducthinh96, 2020-09-04:
I have the exact same problem. My wifi card is AC 8265. I tried the latest stable itlwm kext (1.0.0) and also v1.1.0-alpha.

williambj1, 2020-09-05:
@ducthinh96 Please provide logs and other information

williambj1, 2020-09-12:
Closed due to no further response.

closed 2020-09-12 by williambj1