OpenIntelWireless/itlwm #258: Hotkey toggle wifi

Stevemoretz, 2020-08-15:
**Is your feature request related to a problem? Please describe.**
Making toggling wifi easier.

**Describe the solution you'd like**
Using fn key and stuff seems a bit overhead I guess, I still couldn't get my backlight work with the right combination with fn key. So maybe just a normal and easy keyboard shortcut.

**Describe alternatives you've considered**
My suggestion is simple,in order to this we need 3 commands in terminal, two to turn on and off and one to get the status.That's even more than enough, when you can do it in terminal you can do it with Automator service and set hotkey, piece of cake.

williambj1, 2020-08-15:
Duplicate of #201

williambj1, 2020-08-15:
It’s better to implement this feature in itlwm

closed 2020-08-15 by williambj1