OpenIntelWireless/itlwm #262: Password wrong / forget the network / connection lost every couple of miniutes

soniXx123, 2020-08-16:
**Have You Read Our Docs**
yes

**Are You Reporting A Bug**
yes

**Environment**

- Kext Version: v1.0.0
- WiFi Card Model: Intel Wi-Fi 6 AX201 160MHz
- Product ID: 
- macOS Version: 10.15.6

**Description**
Hi team, I have a couple of problems. 1. my wifi password was wrong and now I don't have the change to change this password because there is no option to change it. As well there is no list of all wifi points you were connected in this case I can't ever forget this network. any idea? I am using now the v1.0.0 and when I have one drive running I lost all the time the connection to the wifi network. 

btw. Safari and AppleTV app is not working as well. Any idea? :) 

**Bug Report Archive**
[bugreport_37014.zip](https://github.com/OpenIntelWireless/itlwm/files/5080600/bugreport_37014.zip)

**Upload your Kext**
[itlwmx.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5080601/itlwmx.kext.zip)

williambj1, 2020-08-16:
Heliport has an option to delete your wrong password; if don’t know how to do so, you can delete them in keychain as well.

Uploading extensively will cause lags, avoid as much as possible.

soniXx123, 2020-08-16:
Ah ok thanks for the replay.

closed 2020-08-16 by williambj1