OpenIntelWireless/itlwm #264: [QUESTION] Does itlwm support WiFi that requires login?

aaronleegit, 2020-08-19:
I'm gonna use a Hackintosh as my daily machine and I am wondering if itlwm supports those WiFi networks with an additional login page, if you understand what I mean. That's because our school has that kind of internet, and I really hope it works with it.

williambj1, 2020-08-19:
Duplicate of #101

closed 2020-08-19 by williambj1

metro2450, 2020-08-24:
From What I read the OP's description, he is not asking about WPA2-Enterprise.  What he wants to discuss seems like a kind of public open WiFi network without password encryption. Once the client connects to the open wifi, the browser will pop up a "login page" to input some kind of credentials like a mobile phone number or email address.

williambj1, 2020-08-24:
> From What I read the OP's description, he is not asking about WPA2-Enterprise. What he wants to discuss seems like a kind of public open WiFi network without password encryption. Once the client [...]

Oops, I've missed that, yes, web page logins are supported.

ErrorErrorError, 2020-08-24:
> From What I read the OP's description, he is not asking about WPA2-Enterprise.  What he wants to discuss seems like a kind of public open WiFi network without password encryption. Once the client [...]

For the most part Universities with WiFi asks for WPA2-Enterprise credentials. The ones that usually show a Browser pop up are the ones with open network however those usually show a reach ability issue, thus the Wi-Fi warning icon on HeliPort.

I can see if I can implement a "login browser" pop up for those that have open internet but requiriese browser authentication.

longyklee, 2020-08-29:
Great! Thanks for the replies and @metro2450 's understanding of my question!