OpenIntelWireless/itlwm #268: Missing BSD Licence header in iwx driver code

stspdotname, 2020-08-24:
The file https://github.com/OpenIntelWireless/itlwm/blob/c859319/itlwmx/itlwmx.cpp contains code derived from OpenBSD's iwx driver and OpenBSD's net80211 wireless stack. Which is fine -- everyone is free to re-use this code for any purpose.

What is not good is that BSD license headers from the original files have been removed. These licence headers should be added back to provide proper attribution and declare all applicable copyrights.

Note that the BSD license says: " [...] provided that the above copyright notice and this permission notice appear in all copies."  This means copyright headers must be carried along with code into any derived files (and files derived from derived files, etc.)

A copy of the original iwx driver code with copyright headers at the top can be found on Github here: https://github.com/openbsd/src/blob/master/sys/dev/pci/if_iwx.c

For example, in the file https://github.com/OpenIntelWireless/itlwm/blob/c859319/itlwm/fw.cpp the iwm driver copyright notice was retained, which is fine. It should be done in the same way for iwx driver code.

Cheers! This is a great project. If you spot any bugs while porting code from OpenBSD, please let me know.

Regards,
Stefan

zxystd, 2020-08-25:
Hi Stefan,
Thank you for figured out my missing mistake, and also thanks for you and your team make this two(iwm and iwx) great projects. Now I have update the licence, you can check it here c6dc7f5.
Also it is my pleasure to report bugs if it is really exists.

Regards,
zxystd

zxystd, 2020-08-25:
fb36789 here is a bug I recently found. The `sc_rx_ba_sessions` value will persist even we associate to another ssid, this will cause sending RX addba refuse frame while we want to start RX agg. reset this value when associated fix this problem. @stspdotname

stspdotname, 2020-08-25:
Thanks a lot! Looks great now.
And indeed, sc_rx_ba_sessions is never reset upon restart. I will try to fix this in our version, too. Thank you :)

zxystd, 2020-08-25:
my pleasure.

stspdotname, 2020-08-25:
Proposed OpenBSD fix for the bug you pointed out: https://marc.info/?l=openbsd-tech&m=159834537103018&w=2

zxystd, 2020-08-25:
can't open the link, but i believe it has been fixed, just a little issue :)

closed 2020-09-19 by williambj1