OpenIntelWireless/itlwm #269: can't connect with iPhone personal hotspot

manmassimo, 2020-08-24:
**Have You Read Our Docs**
yes

**Are You Reporting A Bug**
yes

**Environment**

- Kext Version:itlwm1.0.0 
- WiFi Card Model:ac7235 
- Product ID: itlwm
- macOS Version: 10.15.6

**Description**
doesn't work with iPhone hotspot
**Bug Report Archive**

[itlw.pdf](https://github.com/OpenIntelWireless/itlwm/files/5119612/itlw.pdf)

**Upload your Kext**
[itlwm_v1.0.0_stable.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5119558/itlwm_v1.0.0_stable.kext.zip)

hesongtao96, 2020-08-25:
Same issue.

zxystd, 2020-08-25:
what's the ssid, wifi encryptions? there is only contains scanning logs.

manmassimo, 2020-08-25:
Only that I have. The ssid is isevenup . No more information about it.

williambj1, 2020-09-12:
Closed due to no further response.

closed 2020-09-12 by williambj1