OpenIntelWireless/itlwm #272: itlwm will render IntelBluetoothWireless useless if loaded from OpenCore

tommyvct, 2020-08-26:
**Have You Read Our Docs**

**Are You Reporting A Bug**

**Environment**

- Kext Version: 1.0 stable
- WiFi Card Model: AC3168
- Product ID: 
- macOS Version: 10.15.6

**Description**

If `itlwm` is loaded with `IntelBluetoothFirmware` kexts from OpenCore bootloader, `IntelBluetoothFirmware` will be failed to upload the firmware to the bluetooth part of the wireless card, cause the bluetooth not available for use.
However, if `itlwm` is loaded with `kextutil` the bluetooth will work just fine.

**Bug Report Archive**

[kextutil, no IBF](https://github.com/OpenIntelWireless/itlwm/files/5126855/bugreport_49151.zip)
**Upload your Kext**

[itlwm_v1.0.0_stable.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5126854/itlwm_v1.0.0_stable.kext.zip)

Drun555, 2020-09-01:
I'm using IntelBluetoothFirmware, itlwm and OpenCore, and it is all working together just fine. Btw, I'm experienced same lags before, but with new versions it's becoming a lot better. Didn't have any problems with last releases yet. Did you tried alpha-releases?

zxystd, 2020-09-03:
Tested on Intel 3165ac, no problems you described can be reproduced. I think it is an EFI related issue, maybe you can try to turn to check the EFI or the network card? Thanks.

tommyvct, 2020-09-03:
https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/177

I also filed another issue on IntelBluetoothWireless side, my later thought is this kext is not the culprit, IntelBluetoothWireless is. 

I assume there are some racing conditions that happens by chance.

Also, I found that cleaning the kext caches is particularly important during the process of swapping kexts, but this methodology is not in the troubleshoot section of the document.

williambj1, 2020-09-03:
> Also, I found that cleaning the kext caches is particularly important during the process of swapping kexts, but this methodology is not in the troubleshoot section of the document.

Unless you are installing the kexts in L/E or S/L/E, rebuilding(If it's not the same as “cleaning” plz correct me) kext cache makes no difference at all. Bootloader injection will not be affected unless system dependencies are broken and require some sort of rebuilding to repair; but that has nothing to do with our kexts themselves.

I don't think it's a design flaw that caused a race condition since most of the people didn't experience it. Maybe you can try other's EFI and see if there's any difference.

tommyvct, 2020-09-03:
You are right, I did try load these things up from /L/E. Kext caches rebuilding is required after any change is posted to kexts in /L/E, failure to do so will result in no changes at all. I still highly recommend put this note in the documentation.

alex00x6, 2020-09-07:
For some reason my bluetooth works on Catalina without IntelBluetoothFirmware, via itlwmx only (intel ax200). However, sometimes after booting in Catalina I have issues when macOS and Windows won't see my bluetooth adapter at all, which can be resolved only via some strange shutdown-resetnvram-bootwindows sequence. Not sure how it works yet, thought it might be helpful for the discussion.

tommyvct, 2020-09-14:
I recently updated opencore to 0.6.1.

I injected these kexts with opencore with this order:
[image]
Both the bluetooth and `itlwm` worked without any problem.

closed 2020-09-14 by williambj1

originalpaingod, 2021-01-03:
is there a way to install the itwlm.kext without reinstalling catalina? sorry for the n00b question but it's my first hackintosh (T450s)

matheustavaresdev, 2021-04-07:
@tommyvct did you need to do any other steps or just load the kext files?