OpenIntelWireless/itlwm #273: Itwlm not working when upgrading Opencore

PaulMagos, 2020-08-26:
HI there,
I was using the 0.5.8 version of NDK-OpenCore fork and itwlm used to work really well, today when I was trying to update to 0.6.0 (the official fork), heliport on start said "itlwm not loaded".

What do I have to send to you for helping me ? 

thanks

williambj1, 2020-08-27:
I’m using OpenCore 0.6.1 compiled by myself last week and itlwm works fine. Based on your description, I believe it’s a user configuration mistake, could you please check your EFI?

garretth98, 2020-08-27:
I am having the same problem on OpenCore 0.6.0

ErrorErrorError, 2020-08-27:
> I am having the same problem on OpenCore 0.6.0. I don't see a 0.6.1 release, typo? 

Building the lastest OpenCore gives you 0.6.1

garretth98, 2020-08-27:
Ah okay. Also when I use kextutil to manually load itlwm.kext it works fine, but if I restart I have to manually load it again.

ErrorErrorError, 2020-08-27:
> Ah okay. Also when I use kextutil to manually load itlwm.kext it works fine, but if I restart I have to manually load it again.

Ass @williambj1 mentioned, it's an issue with your efi. Make sure it is enabled in your configure.plost settings.

garretth98, 2020-08-27:
> > Ah okay. Also when I use kextutil to manually load itlwm.kext it works fine, but if I restart I have to manually load it again.
> 
> Ass @williambj1 mentioned, it's an issue with your efi. Make sure it is enabled in your configure.plost settings.

Yeah, my bad. Forgot to add it to config.plist. Everything's working fine now thanks for the help :)

williambj1, 2020-08-27:
Glad that it is solved

closed 2020-08-27 by williambj1