OpenIntelWireless/itlwm #275: [Question] (Posible bug?) TimeOuts and high ping randomly in Network 2.4ghz and 5ghz

Microvivi, 2020-08-31:
I have itlwm 1.0 version, with open core and heliport (Mojave) My card wifi is intel 9260 (I have too 8260 and the problem is the same) I have timeouts and high ping/latency randomly in 2.4ghz and 5ghz.

I Attach screens where u can see the problem.

need u more data?

[image]
[image]

williambj1, 2020-08-31:
Could you provide itlwm logs when this happens? It's not reproducible on our cards. Also please follow the issue template so we could easily identify the problem.

Microvivi, 2020-08-31:
> Could you provide itlwm logs when this happens? It's not reproducible on our cards. Also please follow the issue template so we could easily identify the problem.

Yeah. I attach logs itlwm. Heliport is fine, I can't see errors in logs Heliport.

[itlwm log.txt](https://github.com/OpenIntelWireless/itlwm/files/5150269/itlwm.log.txt)

shanyanhai, 2020-09-05:
heliport1.0.1,itlwmx1.0,macos10.15.6,Router:HONORx2
5Ghz
[image]
2.4GHZ
[image]

log：
[bugreport_29816.zip](https://github.com/OpenIntelWireless/itlwm/files/5178082/bugreport_29816.zip)

williambj1, 2020-09-26:
[image]

Tested 500 pings, no time outs. There are random fluctuations due to background network activity which is expected.

---

@shanyanhai There is nothing wrong in your logs, your issue is more likely related to interferene and local network quality.

---

@Microvivi You are using network extensively according to your logs, pings are expected to fluctuate when tx queues are full. You are also expriencing a ton of output errors caused by a compatibility issue with your router, try adjusting mtu values / channels etc.

closed 2020-09-26 by williambj1

dvrkoo, 2021-01-27:
hello I'm having this issue too has anyone found a fix?