OpenIntelWireless/itlwm #279: [WOW] AirPlay

longyklee, 2020-09-01:
### WOW
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Not really

**Environment**

- Kext Version: 1.0
- WiFi Card Model: Intel Wireless 8265
- macOS Version: Catalina 10.15.6

**Description**
Airplay, surprisingly, works! It works flawlessly with my school's AppleTV! I can't believe this. Because itlwm doesn't really have support for AirPlay, I categorize this as a "bug." I know it shouldn't have worked because itlwm emulates an ethernet device, but I swear it does.

### WOW

JunkuiZhang, 2020-09-01:
interesting

yw0822, 2020-09-01:
unbelievable

zxystd, 2020-09-03:
hmmm, it is an expected behavior, AirPlay can work whether you use WiFi or not, Ethernet interface is OK. Thanks for your feedback.

marras93, 2020-09-03:
Yep I can confirm, it even works on my fake cheap chromecast clone with airplay enabled.

zxystd, 2020-09-09:
:)

closed 2020-09-09 by zxystd

ben001109, 2024-10-29:
@aaronleetw 你是怎麼做到的
re:arronleetw How'd you make it, could you tell me how to configure it? or how can i watch my wifi adapter support or not?