OpenIntelWireless/itlwm #280: [Bug] Disconnecting and reconnecting to WiFi

Le0xFF, 2020-09-03:
**Describe the bug**
HeliPort keeps disconnecting and reconnecting to WiFI after a couple of seconds.

**To Reproduce**
Steps to reproduce the behavior:
1. Open HeliPort
2. Connect to WiFi
3. It will automatically disconnect and reconnect

**Expected behavior**
To keep my connection stable

**Environment (please complete the following information):**
 - OS Version: macOS 10.15.6
 - Software Version: [latest build (2020-08-09) taken from here](https://github.com/1hbb/OpenIntelWireless-Factory/releases)

**Additional context**
My WiFi card is Intel AC 9260 [8086:2526].
I placed kext in "Others" folder of my Clover EFI partition.
I placed HeliPort in my "/Application" folder.
Everytime the app connects to my wifi, it stays connected for a couple of seconds, then it disconnects and reconnects again and again, making my wifi unusable.
I attached also a log, hoping it will help.

[log.txt](https://github.com/OpenIntelWireless/HeliPort/files/5048269/log.txt)

ErrorErrorError, 2020-09-04:
> I am trying to create the debug report and I don't get a prompt to save it?  But I'm having the same issues with strange connection and disconnection.

It saves to your desktop automatically if you click on "Create Diagnostics Report...". Depending on the logs it might take a while to capture all of it. A file window will show once the capture of the logs is complete.

zxystd, 2020-09-09:
fix int 6ce19ee, you can try the latest version.

closed 2020-09-09 by zxystd