OpenIntelWireless/itlwm #283: Random kernel panic triggered by itlwmx

juniorzhou, 2020-09-05:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version:  v1.0.0_stable
- WiFi Card Model: ax200
- Product ID: 
- macOS Version: 10.15.6

**Description**
From the panic log, it looks like the panic was triggered by the kext. The panic will freeze the system. Hard to tell what triggers this panic. I'm also using the Heliport. 
Here are two the panic logs.

[Panic_log1.txt](https://github.com/OpenIntelWireless/itlwm/files/5178435/Panic_log1.txt)

[Panic_log2.txt](https://github.com/OpenIntelWireless/itlwm/files/5178436/Panic_log2.txt)

**Bug Report Archive**

[bugreport_53005.zip](https://github.com/OpenIntelWireless/itlwm/files/5178429/bugreport_53005.zip)

**Upload your Kext**
It the release version of the itlwmx v1.0.0

zxystd, 2020-09-09:
How to reproduce it?

juniorzhou, 2020-09-09:
It's unclear to me how to reproduce. I had those two same panic in the past week. Is there anything you can tell from the panic log or bug report?  I will update this issue when I figure out what cause it.

zxystd, 2020-09-09:
[image]
it is a panic trigger by driver self, because STA can not be remove for twice times, but it is impossible to get here twice times because here have the flag check.

closed 2020-09-19 by williambj1