OpenIntelWireless/itlwm #285: "Hackintool" can no longer retrieve the latest itlwm version

mackonsti, 2020-09-08:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Possibly, I don't know if this is a bug, yet.

**Environment**
- Kext Version: 1.0.0 Stable
- WiFi Card Model: Intel Wireless-AC 9560 Controller
- Product ID: 0x9df0
- macOS Version: Mojave 10.14.6

**Description**
Hackintool by @headkaze was detecting at some point the latest available build for **itlwm** but it seems now we only get a $MARK-something instead. This could be either due to a missing tag in the **Info.plist** of **itlwm** or some change in the repository metadata or even a bug in Hackintool -- only @zxystd and @headzake may know.

**Bug Report Archive**
N/A

**Upload your Kext**
N/A

**Screenshot**

[image]

Polling @headkaze as it could only be a Hackintool issue, I cannot know. Thank you all.

mackonsti, 2020-09-25:
Hi @zxystd can I ask you, when you have some time to reply, if the version tags in GitHub metadata, may need something more to add? The middle column (if you never used Hackintool) is obtaining the latest available versions by checkin on-line each kext on its own Git repo. As I cannot expand the columns, this **$(MARK...)** text is not known to me. Perhaps @headkaze can tell the missing tag?
Thank you!

alessiomatricardi, 2020-09-25:
Now with AirportItlwm, no information is retrieved
[image]

mackonsti, 2020-09-25:
Hi @alessiomatricardi that is because Hackintool was not updated to look for "AirportItlwm" kext, you need to report this and request to be added in https://www.insanelymac.com/forum/topic/335018-hackintool-v3xx/ instead ;-)
I had requested the same for "itlwm" too.

PR #331 (2020-10-07): Use MacKernelSDK & Fix up Project
- Use MacKernelSDK
- Fix Airport kexts build path missing `Products`
- Fix compilation error "FwBinary.cpp" missing
- Fix up versioning, closes #285 
- Clean up codesign
- Add missing `OSBundleRequired` = `Network-Root` in AirportItlwm/info.plist
- Add `ARCHS` = `x86_64`, closes #312 
- Silent `register` deprecation warning
- Update CI
commit 319dd9fc (2020-10-07): Use MacKernelSDK & Fix up Project (#331)

Use MacKernelSDK
Fix Airport kexts build path missing Products
Fix compilation error "FwBinary.cpp" missing
Fix up versioning, closes #285
Clean up codesign
Add missing OSBundleRequired = Network-Root in AirportItlwm/info.plist
Add ARCHS = x86_64, closes #312
Silent register deprecation warning
Update CI
closed 2020-10-07 by zxystd