OpenIntelWireless/itlwm #286: Request suspect firmware

thefiredragon, 2020-09-11:
Dear all, thanks for your work here, first,

I had try your driver for my new laptop, before I misunderstand something I think I have a built in Intel AX 201 Device with the device 0x34F0 Killer Wifi AX1650i 160MHz Wireless Network listed under the Hackintool,

On verbose boot I can see the error | suspect firmware with an number, I could try to provide later my boot log, but first It would be helpful if this card is supported or not, never heard that killer wifi is a branded intel device,
best regards, Dan

ErrorErrorError, 2020-09-11:
Form the docs it is supported. https://openintelwireless.github.io/itlwm/Compat.html#itlwm-kext.

Also, from what I know Intel bought Killer Wireless, and they most likely use Intel chips plus the hardware of River Networks to enhance latency.

williambj1, 2020-09-12:
0x34F0 is supported and we have this id in the kext.

thefiredragon, 2020-09-12:
Thanks for the answer, for me its badly not working, 
: IntelFirmware: Bootloader revision 0.4 build ß well 11 2ß17IntelFirmware: suspect device firmware: ibt-19-32-4.sfi

Im using itlwmx.kext any ideas? Perhaps this device has an unsupported firmware?

williambj1, 2020-09-12:
I don't see any issue in your logs; the kext is already scanning networks. Maybe you missed something? Have you read our docs?

thefiredragon, 2020-09-12:
yes I missed up heliport application to install and use selecting networks, yet it's working fine, thanks all for this work and sorry for my mishandling here,
Dan

closed 2020-09-12 by thefiredragon