OpenIntelWireless/itlwm #288: [bug report]itlwm在自动熄屏后断网

daimali666, 2020-09-12:
**Have You Read Our Docs**
是

**Are You Reporting A Bug**
是

**Environment**

- Kext Version: commit 832cf1aa5d44deec4724dbfaa4c644282f3da4ea（截至9/12的最新版，精简过固件）
- WiFi Card Model: CNVi 8265AC
- Product ID: 0x24fd
- macOS Version: macOS Catalina 15.6
- HeliPort Version: V1.0.0 (24cfae7，stable版)

**Description**
我的itlwm在**自动熄屏**（而不是睡眠）中断网，heliport无法重新打开wifi（Wi-Fi状态显示“关闭”，且点“开启Wi-Fi“无反应”），我应该怎么办？
**Bug Report Archive**
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5212867/itlwm_logs.log)

**Upload your Kext**
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5212868/itlwm.kext.zip)

**Additional context**
我的电脑：HUAWEI Magicbook 2018 (i5-8250U)
熄屏断网的截图：
[image]

zxystd, 2020-09-14:
PCI device operation fail, due to EFI issue, you should fix it by yourself. Thanks.
[image]

closed 2020-09-14 by zxystd

uywolfer, 2020-11-23:
big sur也有这种问题同款机型，

uywolfer, 2020-11-23:
我的解决了，系统版本是11.1，用airportitlwm1.2，然后config里面删掉pci-aspm-default=o

uywolfer, 2020-11-23:
我的解决了，系统版本是11.1，用airportitlwm1.2，然后config里面删掉pci-aspm-default=o，群里试过的都解决了

Zero-zer0, 2020-11-23:
The problem was solved by removing `pci-aspm-default=0` under device property of Intel wireless cards, according to some people who get the same issue with the same EFI confirgration.

Such property is used to disable ASPM for BCM94350/ DW1820A to avoid booting issues. However, it can now be patched by `AirportBrcmFixup` automatically for BCM94350/ DW1820A. Also, such property can be injected by SSDTs. For example: [OC-little](https://github.com/daliansky/OC-little/blob/4327527dbbc9942c5d8483c1b8d8a1a3ee5322d8/01-%E5%85%B3%E4%BA%8EAOAC/01-5-%E8%AE%BE%E7%BD%AEASPM%E5%B7%A5%E4%BD%9C%E6%A8%A1%E5%BC%8F/README.md)

Maybe we can remind user in README to remove all PCI property injection of the wireless card.