OpenIntelWireless/itlwm #290: The Wi-Fi display is not configured and cannot be turned on.

CLAY-BIOS, 2020-09-16:
Have You Read Our Docs
yes

Are You Reporting A Bug
yes

Environment

Kext Version: AirportItlwm.kext
WiFi Card Model: Intel AX 200
Product ID: 0x2723
macOS Version: 11.0 beta6
Description
Wi-Fi icon exclamation point,The Wi-Fi display is not configured and cannot be turned on.

Bug Report Archive
N/A

Upload your Kext
N/A

Screenshot
[image]

zxystd, 2020-09-17:
How to reproduce?

lgariv, 2020-09-17:
> How to reproduce?

Just happened to me as well on Catalina.
Put it in OpenCore (0.6.1) kexts folder, added to config, and reboot.

after loading it from terminal, it was working - but not ideal for everyday use and without handoff/clipboard continuity support (at least not working for me).

lgariv, 2020-09-17:
> > How to reproduce?
> 
> Just happened to me as well on Catalina.
> Put it in OpenCore (0.6.1) kexts folder, added to config, and reboot.
> 
> after loading it from terminal, it was working - but not ideal for everyday use and without handoff/clipboard continuity support (at least not working for me).

Update: 
In my case, setting SecureBootModel to Default and ApECID to 0 fixed the issue. might be an issue with OpenCore.
Handoff and Clipboard Continuity still does not work tho.

williambj1, 2020-09-17:
https://github.com/OpenIntelWireless/itlwm/issues/292#issuecomment-693728787

lgariv, 2020-09-17:
> #292 (comment)

Tried that, didn't work.

williambj1, 2020-09-17:
>  Also, not sure if that's a user mistake as the person opened this issue is on Big Sur and I'm on Catalina,

I will remove the label for now and reconfirm it later, thanks for pointing out.

> Tried that, didn't work.

Usually it takes some time for the service to initialize and make sure you followed all of the steps. Handoff it a bit off-topic here, maybe we can discuss further in #292 or Gitter.

CLAY-BIOS, 2020-09-17:
I found the reason.
When I use itlwm.kext, in order to display real-time internet speed on third-party software,
I used this method #172 to establish a bridge.
When I used AirportItlwm.kext, the bridge was not deleted which caused this problem.
After deleting the bridge, it returns to normal.

williambj1, 2020-09-17:
Glad that it's fixed :)

closed 2020-09-17 by williambj1