OpenIntelWireless/itlwm #291: AirportItlwm似乎与蓝牙有冲突

ske1996, 2020-09-16:
**Have You Read Our Docs**
是的

**Are You Reporting A Bug**
是的

**Environment**

- Kext Version: v1.1.0 stable release中的AirportItlwm_0916_BigSur.kext.zip
- WiFi Card Model: 第一代ac9560
- Product ID: 0x8086, 0x2034
- macOS Version: BigSur public beta 1

**Description**
在使用itlwm时不曾出现，于今日更换AirportItlwm_0916后出现蓝牙无法连接设备，一旦连接设备会在偏好设置-蓝牙位置卡死，在终端中输入【log show --last boot | grep IntelFirmware】无法获取log
确定usb的定制没问题，确定加载了蓝牙的kext，确定bios中没有ban掉蓝牙

为了开机秒连，我在AirportItlwm中写入了ssid 跟password

蓝牙驱动版本为8/4日的1.12

附件添加引导日志+系统日志+两个驱动
[引导日志.txt](https://github.com/OpenIntelWireless/itlwm/files/5235261/default.txt)
[系统日志.txt](https://github.com/OpenIntelWireless/itlwm/files/5235262/default.txt)
[归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5235266/default.zip)

zxystd, 2020-09-17:
@williambj1 你的网卡也是一样的，有没有这个问题？

ske1996, 2020-09-17:
> @williambj1 你的网卡也是一样的，有没有这个问题？另外是什么蓝牙设备？

连接airpods与iPhone时均出现此问题

ske1996, 2020-09-17:
连接iPhone是为了使用一个叫做虫洞的的app，将iPhone投屏到mac用，蓝牙作为电脑上操作iPhone的通道

williambj1, 2020-09-17:
Known issue.

ske1996, 2020-09-17:
目前情况：更新macos至dp6后蓝牙有时可用了，可以使用AirPods了，但是我重启了6次，只有一次可以连接上，为什么可以连接上（或不可以）实在不明，不知道有没有什么需要我提供的，本次重启后handoff可用，airdrop不可用，airpods可以连接并播放。

附件为本次上述蓝牙连接设备（服务）可用后的日志
[引导日志.txt](https://github.com/OpenIntelWireless/itlwm/files/5235776/default.txt)
[系统日志.txt](https://github.com/OpenIntelWireless/itlwm/files/5235777/default.txt)

williambj1, 2020-09-17:
我能连上 Airpods 的概率为 1/10，应该和 iCloud 的同步配对有关。

ske1996, 2020-09-17:
非常感谢二位做出的贡献

ske1996, 2020-09-17:
补充: handoff可以在iphone复制后mac上粘贴。但是反着不行

ske1996, 2020-09-17:
再次补充:如果在偏好设置里关闭wifi，或者wifi没有联网的情况下，蓝牙就没什么问题

场景二:关闭wifi后可以顺畅连接airpods并播放音乐等，但是只要一旦开启wifi去连接上热点后，蓝牙就会被立刻强迫掐断

tommyvct, 2020-09-17:
你opencore里注入的顺序是什么？
按照[这个顺序](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/177#issuecomment-691855028)试一下?

ske1996, 2020-09-18:
> 你opencore里注入的顺序是什么？
> 
> 按照[这个顺序](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/177#issuecomment-691855028)试一下?

好的谢谢你，我会去试的

ske1996, 2020-09-18:
> 你opencore里注入的顺序是什么？
> 
> 按照[这个顺序](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/177#issuecomment-691855028)试一下?

谢谢你，按你说的已解决

tommyvct, 2020-09-20:
@zxystd 你们有专人在负责文档工作么？要不要把这个载入顺序写道文档里？

williambj1, 2020-09-20:
> @zxystd 你们有专人在负责文档工作么？要不要把这个载入顺序写道文档里？

Will do

Currently available at https://github.com/daliansky/OC-little/blob/master/常见驱动加载顺序/config-6-Intel无线和蓝牙驱动列表.md

ske1996, 2020-09-21:
emm, i got that issue again...i am sure i didnt changed anything since i thought i fixed wifi-bluetooth conflict issue.

today,I got that again,I cant connect to any bluetooth device while wifi is on.(bluetooth works fine by turning wifi off)

here is my opencore config.plist.

thank you again.

[config.plist.zip](https://github.com/OpenIntelWireless/itlwm/files/5254768/config.plist.zip)

Kyleinhurricane, 2020-10-07:
我一般暂时把wifi关上之后AirPods就可以连上了，然后开关几次wifi有时可以比较稳定的wifi和蓝牙同时使用。

dolame, 2020-10-10:
开启蓝牙和wifi蓝牙传输受影响，蓝牙耳机卡顿

vimcaw, 2020-11-05:
> 你opencore里注入的顺序是什么？  
> 按照[这个顺序](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/177#issuecomment-691855028)试一下?

奇怪了，我这里正好相反，itlwm 在上面会出问题，一直搜不到设备或者连接不上，把 itlwm 放到最下面是正常的 🤣。

vimcaw, 2020-11-05:
[image]

我这里要把 AirportItlwm.kext 放到蓝牙下面加载才能正常使用，放到蓝牙上面就会搜不到设备，之前配对过的也连接不上。

zncs520, 2021-01-12:
7265AC 按照顺序 正反都一样不行

szhielelp, 2021-01-17:
报一个奇怪的问题, 7265AC
昨天AirportItlwm.kext + 2个蓝牙的运转良好, 蓝牙和WIFI都正常
今天蓝牙的2个怎么换位置都不行

llaisyuan, 2021-02-25:
不知道我发的会不会有人去解决或观注
现在11G系统OC6.6 网卡AX200  
 我的情况比较特殊：WIFI都能正常使用，蓝牙不管处于什么状态都不工作
1：WIFI和蓝牙驱动都在OC中加载，那么 WIFI正常使用 蓝牙在USB中看不到，且蓝牙设备属性为空
2: 只加载WIFI驱动，那么USB中不会出现蓝牙，但蓝牙属性中会看到蓝牙这个设备，厂商识别为爱立信
3：不管任何状态，放入蓝牙驱动，那么USB中和蓝牙属性中都看不到蓝牙设备
4：如果驱动都不放，那么蓝牙在USB中会出现，且蓝牙设备管理中可以看到它
USB端口我也定置过，我在没有WIFI的状态下，能看到蓝牙USB端口的时候去定置了，然后重新加载WIFI和蓝牙驱动也没有用
请求帮忙，或给出建议

QQ：973264

tommyvct, 2021-02-25:
> 不知道我发的会不会有人去解决或观注
> 现在11G系统OC6.6 网卡AX200
> 我的情况比较特殊：WIFI都能正常使用，蓝牙不管处于什么状态都不工作
> 1：WIFI和蓝牙驱动都在OC中加载，那么 WIFI正常使用 蓝牙在USB中看不到，且蓝牙设备属性为空
> 2: 只加载WIFI驱动，那么USB中不会出现蓝牙，但蓝牙属性中会看到蓝牙这个设备，厂商识别为爱立信 [...]

可能跟主板有关系，IRQ可能有冲突，aml没配置到之类的

llaisyuan, 2021-02-25:
但我无

> 可能跟主板有关系，IRQ可能有冲突，aml没配置到之类的

但我无从下手，不是太懂，哎！

win1010525, 2021-02-25:
> 不知道我发的会不会有人去解决或观注
> 现在11G系统OC6.6 网卡AX200
> 我的情况比较特殊：WIFI都能正常使用，蓝牙不管处于什么状态都不工作
> 1：WIFI和蓝牙驱动都在OC中加载，那么 WIFI正常使用 蓝牙在USB中看不到，且蓝牙设备属性为空
> 2: 只加载WIFI驱动，那么USB中不会出现蓝牙，但蓝牙属性中会看到蓝牙这个设备，厂商识别为爱立信 [...]

Invalid as you never load IntelBluetoothFirmware normally.
Off topic @williambj1

closed 2021-07-18 by ske1996